phrack/ShootOFF #424: Implement distance perspective handling

cbdmaul, 2016-03-30:
Follows conversation started in #423 

Implement a way to input, calculate, and maintain targets based on real world sizes.

I intend to implement it as described.

> OK, it seems there are multiple requirements here we should tease apart in a separate issue (I'll let you open it since you seem to be claiming it -- which I am OK with):
> 
>     Each target must be capable of appearing at a unique distance
>     It must be possible for the projector or shooter to change their distance from all targets via user input at any time (I think it's safe to assume that the projector distance -- from camera to projector -- is always the shooter distance for simplicity).
>         This change must be simultaneously reflected in all targets (e.g. all targets get bigger immediately if user is closer)
> 
> Based on that it seems like you want each target to know its distance in yards and its scale (it already knows its scale). When a target has its distance set, it is registered with a PerspectiveManager if it has not been registered yet (deregistered if this distance is deleted by the user?). The PerspectiveManager knows the shooter's distance and is responsible for telling newly registered targets what its scale should be based on both the shooter and the target distance (the target less the shooter distance?). If the shooter distance is changed it is updated in the PerspectiveManager that knows all targets with set distances and can therefore update them all.
> 
> PS. The real point here is that the Target probably shouldn't know anything about how or when to scale itself, it should just know the metadata completely unique to it (its distance and its scale due to that distance). Some other component should know how to scale any target and when to scale them.
> 

Should the target metadata be baked into Target or should I make some kind of perspective object that is associated with every target?

phrack, 2016-03-30:
I think the metadata can be baked into the target implementation class (TargetView) but the getters and settings for things like scale and distance should be defined in an interface specific to perspective metadata. The PerspectiveManager shouldn't get passed the full target, it should just get passed the metadata interface version (e.g. PerspectiveMetadata). This way we reduce coupling between the perspective code and target code and we make it easier to refactor those getters and setters parameters and return types later if we need to (e.g. change their definitions in the interface, then left the type errors from the compiler tell you where to go refactor).

My logic here is that target distances are very commonly expressed in feet or yards, thus you don't need a fancy interface right now. You basically need:

    public void setDistance(float distance /* in yards, e.g. 1.5 = 3 feet 18 inches */);
    public float getDistance();
    public void setScaleX(double scaleX); // implementation can just call targetGroup.setScaleX(scaleX) in TargetView
    public double getScaleX(); // targetGroup.getScaleX() -- super easy
    public void setScaleY(double scaleY);
    public double getScaleY();

Now if we decide to do something more complicated because we need units more precise than we will reasonably get with a float (e.g. inches) for, say, the precision calculator, we replace the float in (set|get)Distance with some object type (e.g. Distance) that lets us express more specific and clearly defined units. But since we don't know if we actually do need that until we design a precision calcultor, let's not complicate yet. The interface will make it super easy to refactor later if we really need to.

cbdmaul, 2016-03-31:
Plugins may need access to perspective information, don't be such a plugin downer

phrack, 2016-03-31:
We should be very careful about what the use case is and generally should be downers about exposing things to plugins unless there is a very compelling reason. Now that we have modular plugins, once 3.8 is out we have to assume people are using anything we document as plugin API. All of those documented points are stuff we now have to be extremely conservative about changing to minimize breaking plugins. 

At the very least, I'd prefer to let something like this evolve for a few releases before we let plugins use it.

cbdmaul, 2016-04-01:
I started working on it.  Two observations

1. I need to whiteboard this because the math is very confusing
2. It may just be the ugliest class I've ever written.

cbdmaul, 2016-04-26:
Okay, it is working.  There is an outstanding issue:

It works if the feed resolution is the aspect ratio that matches the sensor's native ratio.  For instance, the C270 is 16:9.  All 16:9 feed resolutions work such as 1280x720.  However, if it is 640x480, which is 4:3, then the math doesn't work out.  The sensor width and height is very clearly based on the aspect ratio: 3.58 / 16 * 9 = 2.02.  So I could correct it for this camera in theory, but is this a general thing?

cbdmaul, 2016-06-25:
Seems like everything is working on my end.  Need to integrate into exercises, add dialogs/settings where appropriate.

phrack, 2016-06-28:
- [ ] Add the perspective calibration target as a ready-to-print PDF to the targets folder.
- [ ] The .target files for most of the targets need a default realWidth and realHeight in mm and the corresponding camera distance in mm (e.g. competition target that is 36" x 24" at 10 yards). 
- [ ] When a target with default data is added to the projector arena with known perspective parameters, automatically resize the target appropriately.
- [ ] Let the user set the default target size and distance in the target editor. This needs to be a general target property but currently properties (tags) can only be added to individual regions.
- [ ] Add a distance dialog opened by right clicking a target and hitting "Set Target Distance", with an OK button to accept the settings and resize the target and a cancel button to give up:
    - [ ] Always show the default width, height, and distance; these parameteres MUST be filled in when OK is hit or user gets a popup error
    - [ ] Always ask for the shooter's distance from the camera. If this is not set it defaults to the camera distance. Persist shooter distance per camera in the settings file.
    - [ ] Always ask for a target distance, but default it to the same as the shooter distance (shooter distance defaults to camera distance)
    - [ ] If we have a paper pattern but no camera parameters, ask for the camera distance in mm (shows blank text box), this MUST be filled in when OK is hit or user gets a popup error
    - [ ] If we have the camera distance from settings and either the paper pattern or camera parameters, we don't need to calculate anything so show the camera distance so that it can be overridden
    - [ ] If we have a camera distance and no paper and no camera parameters, show a popup asking user to recalibrate with the paper pattern
    - [ ] Camera distance, if set by the user (i.e. change in value to something different from what it was when the dialog opened), needs to be persisted to the settings file on a per-camera basis.
- [ ] Add a method to the projector API that takes in a target and a target distance. If any parameters are missing (e.g. realHeight, realWidth, camera distance/camera parameters/paper) return false (debug log with what was missing) and do nothing, otherwise return true.
- [ ] Add documentation about how to access the distance dialog, how to fill it in, how to print the pattern paper from PDF with a note about non-glossy paper, and where to place it.
- [ ] Add documentation for projector API method.

cbdmaul, 2016-06-28:
Looks like it works regardless of pattern orientation already.  Confirmed by rotating test image and physical test.  Autocalibration doesn't seem to be getting the paper pattern every time, but not worth a lot of effort to investigate right now.

cbdmaul, 2016-06-28:
I started a wiki page: https://github.com/phrack/ShootOFF/wiki/Using-Precise-Real-World-Distances

phrack, 2016-07-03:
@cbdmaul, I consider the ball to be back in your court on this. We have full integration for regular users and training exercises, plus documentation for both:

https://github.com/phrack/ShootOFF/wiki/Using-Precise-Real-World-Distances
https://github.com/phrack/ShootOFF/wiki/Creating-Training-Exercises (all the way at the bottom)

At this point if there are any bugs or tweaks in how something should work, it's almost definitely faster for you to just do them real quick since perspective projection is your vision.

phrack, 2016-07-06:
Per email: 

You use the result to set the "perceived" values of the target but perceived values are in MM and the return of calculateObjectSize is in projector resolution pixels (How many pixels of the projector's output resolution are needed to show the target at the right size).

			tags.put(Target.TAG_CURRENT_PERCEIVED_HEIGHT, String.valueOf(d.getWidth()));
			tags.put(Target.TAG_CURRENT_PERCEIVED_HEIGHT, String.valueOf(d.getHeight()));

cbdmaul, 2016-07-06:
You're still changing the perceived distance to the desired distance.  The perceived distance should never change.  If you look below, I'm showing the target at a desired distance of 7000.  Then I switch it back to 3503, but because realDistance is 7000 now, I get the same target size I had already.

2016-07-06 19:38:34 [JavaFX Application Thread] TRACE c.s.c.perspective.PerspectiveManager - rD 3503.0 dD 7000.0 sD 3503 dR 0.5004285714285714 - adjmm 139.61957142857142 108.09257142857143 adjpx 70.09740696207852 57.235352262644184
2016-07-06 19:38:48 [JavaFX Application Thread] TRACE c.s.c.perspective.PerspectiveManager - shooterDistance 3503
2016-07-06 19:38:48 [JavaFX Application Thread] TRACE c.s.c.perspective.PerspectiveManager - cameraDistance 3503
2016-07-06 19:38:48 [JavaFX Application Thread] TRACE c.s.c.perspective.PerspectiveManager - rD 7000.0 dD 3503.0 sD 3503 dR 0.5004285714285714 - adjmm 139.61957142857142 108.09257142857143 adjpx 70.09740696207852 57.235352262644184

cbdmaul, 2016-07-06:
Seems like if I just remove that other tag put, I get the behavior I expect

phrack, 2016-07-06:
I was just going to tell you that removing the other tag put is all it should take to fix it.

cbdmaul, 2016-07-07:
There's still a scaling bug.  The math escapes me.

cbdmaul, 2016-07-08:
It's fixed now but GUI elements and such probably still need polish.

Bottom line, I screwed up by distinguishing perceived distance (or realDistance) from desired distance.  They're really the same thing, the target's default distance is really just the initial value for desired distance until the user wants something else.

I feel like you had some idea this wasn't right and I just didn't catch on to it.  I'm willing to accept a "I told you so" here

phrack, 2016-07-08:
Glad to hear you got it. Any specific proposals GUI wise? I am inclined to make the dialog show up over the webcam feed instead of the arena because text entry is really painful with this setup. It already should only show the target width and height if there were not defaults, but that could be buggy. Again, I don't want to spend too much time getting it right since the very next thing we are doing is rethinking the entire GUI.

cbdmaul, 2016-07-08:
I don't see any reason to not show the default target width/height and let it be modified by the user, but it's also such a minor thing and with GUI changes coming it doesn't matter to me.

The GUI can just show the target distance defaulted to the default distance, and then allow the user to change it from there.  I think that's how it is now but since I had to modify the GUI code to fix references I suggest giving it a once over to verify that things are shown at the correct times.

Along those same lines, it would be nice to have a "reset to default" setting in there but once again, that's an improvement that's not really needed at this time.

There is still a condition that needs to be handled: desiredDistance == 0 (or low enough that the target exceeds the size of the screen).  I put a TODO in there

phrack, 2016-07-09:
If for some reason there is actually more to do here (probably GUI tweaks if anything), let's open a separate issue for them. It seems done and this issue is already quite crowded.

commit d97595c0 (2016-07-09): Finish dealing with invalid perspective values to close #424
closed 2016-07-09 by phrack