phrack/ShootOFF #438: Shot to click is clicking on the wrong screen, in the upper right corner

cbdmaul, 2016-04-04:
Title is about all I know.  Even if I move the mouse to the right screen it moves it back to a small region of my primary display in the upper right corner (projector is secondary display on the right of the primary).

It doesn't move it to the same spot every time, it seems to think a small region of my primary display is the entire secondary display.

DPI scaling problem maybe?

phrack, 2016-04-04:
I think this is DPI again. It's working without a high resolution screen present.

phrack, 2016-04-04:
"Java applications still advertise themselves as “system DPI aware,” i.e. they don’t opt into the new per-monitor DPI scaling of Windows 8.1. Support for this feature is planned for Java 9."

http://news.kynosarges.org/2015/06/29/javafx-dpi-scaling-fixed/

Does this seem relevant?

phrack, 2016-04-04:
Ahh, it also could be because AWT doesn't do DPI scaling and we use an AWT class to simulate a click:

https://docs.oracle.com/javase/7/docs/api/java/awt/Robot.html

If this is the problem it means the scaling has to happen in the exercise and not in ShootOFF's code (i.e. this issue is in the wrong place if this is the problem).

cbdmaul, 2016-04-04:
The per-monitor scaling issues could be what made the arena stop opening to the right screen when I updated to 8u60.  It could also cause problems here.

We might not be able to do much about either case until Java 9 in May.  Maybe we should review what their plan is and make sure it actually solves our problems, while it is still in open comment.

phrack, 2016-08-13:
@cbdmaul this needs the DPI fix too.

cbdmaul, 2016-08-15:
This is fixed on the ShootOFF side.  Just waiting on one last change on the Shot To Click exercise

cbdmaul, 2016-08-15:
Confirmed

closed 2016-08-15 by cbdmaul