phrack/ShootOFF #446: Implement a disappearing bouncing targets exercise

cbdmaul, 2016-05-17:
Display a group of bouncing targets.  Remove targets that are hit.  Reset when there are no more targets.

More info: https://www.reddit.com/r/ShootOFF/comments/4jadgr/bouncing_targets/

ifly53e, 2016-05-31:
code to make this happen has been inserted at the "mod" comments below.

/*
 * ShootOFF - Software for Laser Dry Fire Training
 * Copyright (C) 2016 phrack
 * 
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 * 
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 * 
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

package com.shootoff.plugins;

import java.io.File;
import java.util.ArrayList;
import java.util.List;
import java.util.Optional;
import java.util.Random;

import javafx.animation.KeyFrame;
import javafx.animation.Timeline;
import javafx.collections.FXCollections;
import javafx.collections.ObservableList;
import javafx.geometry.Bounds;
import javafx.geometry.Dimension2D;
import javafx.geometry.HPos;
import javafx.geometry.Point2D;
import javafx.scene.Scene;
import javafx.scene.control.Button;
import javafx.scene.control.ComboBox;
import javafx.scene.control.Label;
import javafx.scene.layout.ColumnConstraints;
import javafx.scene.layout.GridPane;
import javafx.stage.Modality;
import javafx.stage.Stage;
import javafx.util.Duration;

import com.shootoff.camera.Shot;
import com.shootoff.targets.Hit;
import com.shootoff.targets.Target;

public class BouncingTargets extends ProjectorTrainingExerciseBase implements TrainingExercise {
	private int shootCount = 4;
	private int dontShootCount = 1;
	private static int maxVelocity = 10;
	private boolean disappearingTargets = false;//mod

	private static final List<BouncingTarget> shootTargets = new ArrayList<BouncingTarget>();
	private static final List<BouncingTarget> dontShootTargets = new ArrayList<BouncingTarget>();

	private static ProjectorTrainingExerciseBase thisSuper;
	private Timeline targetAnimation;
	private int score = 0;

	private boolean testing = false;

	public BouncingTargets() {}

	public BouncingTargets(List<Target> targets) {
		super(targets);
		setThisSuper(super.getInstance());
[... 275 lines cut ...]
		}
	}

	@Override
	public void reset(List<Target> targets) {
		targetAnimation.stop();

		for (BouncingTarget b : shootTargets)
			super.removeTarget(b.getTarget());
		shootTargets.clear();
		for (BouncingTarget b : dontShootTargets)
			super.removeTarget(b.getTarget());
		dontShootTargets.clear();

		if (!testing) collectSettings();

		addTargets(shootTargets, "targets/shoot_dont_shoot/shoot.target", shootCount);
		addTargets(dontShootTargets, "targets/shoot_dont_shoot/dont_shoot.target", dontShootCount);

		targetAnimation.play();

		score = 0;
		super.showTextOnFeed("Score: 0");
	}
}

ifly53e, 2016-06-01:
Pull request added

cbdmaul, 2016-06-01:
Verification of the pull request will have to wait for Phrack's approval, I'm not familiar with the exercise components.  He should be back at it soon.

phrack, 2016-06-01:
I am back, but will take a few days to get to the pull request. I am still working my way through emails (almost done though) and haven't even looked at the subreddit yet.

ksorken, 2016-06-08:
During the tests, bouncing targets2. jar (plugin) I saw a problem with the speed settings. Try setting (10 (speed) dissapear YES, then 1 speed dissapear YES, then 10 dissapear YES) several times. If you set the speed 1 (after setting 10) setting does not work. Speed 1 is faster at first time. Then compare the speed "1" of the original bouncing targets. When you change the speed never restart the program.
(Shootoff 3.8)
Platform:
Linux 3.16.0-4-amd64 Debian Jessie
java version "1.8.0_77"
Java(TM) SE Runtime Environment (build 1.8.0_77-b03)
Java HotSpot(TM) 64-Bit Server VM (build 25.77-b03, mixed mode)

ifly53e, 2016-06-08:
I could not replicate this problem in my setup...

I did notice that speed setting is a maximum so you can't assume all of the targets will travel at the same speed.

   public BouncingTarget(Target target) {
        this.target = target;

        Random r = new Random();

        dx = r.nextInt(maxVelocity + 1) + 1;
        dy = r.nextInt(maxVelocity + 1) + 1;

        if (r.nextBoolean()) dx *= -1;
        if (r.nextBoolean()) dy *= -1;
    }

The above code takes the maximum velocity setting and makes the x and y change in velocity random numbers between 1 and that setting number.  So if you are expecting to see all of the moving targets move at the same speed when the setting is 10, then that will only happen by random chance.

All of the targets should be moving pretty slow when the setting is 1 so I am not sure why you are seeing major changes in speed at the slowest setting.

phrack, 2016-06-08:
I have three suggestions at the moment (perhaps more if there is a pull request):

1. The Yes/No setting for making targets disappear should probably just be checkbox instead of a dropdown list and a label. Unless, of course, there is some good reason for it to not be a checkbox.
2. Targets should be removed when "disappeared" instead of making them invisible. If they are removed they don't need to be processed under the hood anymore. Also, a shot on an invisible target can still be detected. We have several targets with shootable regions with their visibility set to false.
3. This change is a combination of simple enough and useful enough to just incorporate in the official version instead of having a "Bouncing Targets 2" IMO.

ifly53e, 2016-06-09:
Phrack,
1.  Small dog to puppy...not consistent with previous shootOff exercises.  Personal preference and it is your project.  Overkill...yes, but it is done.
2.  If the targets are not drawn in the scene, how can they be hit?  I realize they are still being moved under the hood but that is a small performance hit compared to the search you are going to have to do to find the Target, compare it to the bouncingTarget, and then remove the bouncingTarget from the list all during the shotDetection algorithm.  That is more code to implement than the one liner to make the target invisible.  But, I would like to see how you would do that since I was not about to invest the time for a change I knew you would probably not accept in the first place.
3. I am not a core developer so I would never get that change with a pull request anyway so learning how to complete and compile a module and submitting the pull request according to the rules you put forth in the tutorial was my only option.

I hope that ksorken got some use out of it while he was waiting for Phrack to get to this.

Welcome back, Phrack. 

I missed you.

Love, Ifly53e

phrack, 2016-06-09:
My point about 2 is that the request seems to be asking that the exercise reset once each shoot target is shot at least once (disappearing just being a nice way to know which ones have been shot and remove distractions). That is not what is implemented. An invisible target can still be shot because it's still in the list and the hit detection algorithm doesn't take the visibility of the JavaFX node into account (https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/TargetView.java#L409), only if the target is in the list + the top most target + with a shot in one of its region's bounds (if it's an image region the shot can't be on a pixel whose alpha value makes it invisible -- not the same as JavaFX node visibility). With what's implemented you can hit a shoot target multiple times and the exercise will reset even though you haven't shot each of them once because `score==shootCount`. I shouldn't have brought up efficiency because you're right that it's a red herring when the point is really about correctness. Maybe I am interpreting the request wrong though.

We do accept pull requests from non-core developers and have many times. Yours rarely get accepted  (or so far have never been?) because you seemingly refuse to make corrections to remain consistent with the rest of ShootOFF, make even small quality improvements, etc. The GUI point is a good example: no where else in ShootOFF is a dropdown list used to graphically represent a boolean value. In all cases where a dropdown list is used it's because there is no better way to graphically reflect the input. The speeds for example have to be bounded (e.g. 1 - 10), a text box isn't great for this in this case because the space of possible speeds it not very big and documenting what freeform inputs are actually valid would be burdensome to us and users. 

I am not picking on you. I wish we had enough developers that I could be reviewed too. Even half of my commits would be rejected on the first try because, just like proof reading prose, it's very hard to spot even some obvious bugs in your own code unless you haven't touched it in a few weeks. This would save us a lot of bugs.

phrack, 2016-06-09:
Seriously, on the review point: filter our closed issued by bugs. There are 209 closed bugs. It's a safe bet that this number would be cut in half or more if we had enough developers all along that even the core developers were getting reviewed and rejected.

commit 24aceb21 (2016-06-17): Add option to remove shoot targets that are hit in bouncing targets to close #446
closed 2016-06-17 by phrack