phrack/ShootOFF #447: ShootOff window size when adding columns

ifly53e, 2016-05-25:
After using addShotTimerColumn, the main window should adjust to a size to allow the user to view the additional columns without scrolling until the desktop space is gone.

phrack, 2016-06-08:
I think this issue is going to sit until 4.0 since we will be overhauling the GUI in 4.0 and 3.8 is intended to be the last release in the 3.0 series.

ifly53e, 2016-06-09:
I have a feeling we are going to be on 3.8 for a while and this would be useful for those of us who have already made exercises with several columns of additional information.  But this is small potatoes compared to things like native shot detection...  Would you let a change to the source go through by someone with a solution other than a core developer?  I found the code where the shootOff window is created and just increased the y value...can you at least do that?

How close are we to 3.8 release?

Is there someplace where we can see the roadmap, or the development plan, for the point and major releases?  You don't owe us anything, but then again, you made it public and the user community is completely reliant upon you.  I think about this project a lot, and I have my own fantasies for how I am going to use it.  I would rather be shot down sooner and invest my time elsewhere instead of being surprised by another flip similar to the 3.6/3.7 releases.

phrack, 2016-06-09:
Now, see about the pull request. This is a good example: If you just add to the width value I am going to come back and say, "OK, but what if adding that x value causes the window to grow off the screen? You should add a check to only add to the width if that won't happen." Then if the precedent holds you'll disappear wrt to that set of code. I make oversights like that all the time and **wish** we had enough people to review every bit of code like you're getting reviewed. We've made a lot of improvements in quality in the last year. Reviews would probably double the degree of those improvements in a few versions.

Native shot detection is becoming problematic because the developer who is working on it disappears for long periods of time and then has been popping back up with just enough progress to keep us on the hook. I am very close to the point of saying it won't be in 3.8 and won't even be on the roadmap until 4.0 is out. That'd be a very frustrating thing to do because the native shot detection implementation IS close enough that we've tried it in practice. Another option might be to get the code into a different repository (the developer is doing it in their own separate repo) so we can directly help it along. I just don't really know what else to do and we can't sit on our hands indefinitely waiting for it. 

Our roadmap currently lives in a private issue tracking system where we have many, many different wish list items (some because they are really neat and we don't want someone to steal the idea before we get a chance to try it, some because they require partnerships with organizations in the real world where we don't want to prematurely show our hand). I am open to keeping two milestones open at all times: one for the version we are working on and one for the next version. Is that a reasonable compromise.

ifly53e, 2016-06-10:
Thanks for giving the user community a glimpse at the roadmap.

phrack, 2016-07-16:
Closing since we'll be redoing the GUI very soon given that there is little else to do in 3.8.

closed 2016-07-16 by phrack