phrack/ShootOFF #448: Target size after addTarget or manually

ifly53e, 2016-05-25:
I made a known size target in photoshop that was a 1/3 scale of an actual target.  The original target is 24"x36" but javafx doesn't like pushing around all those pixels so I scaled it to 8"x12".  When adding the target to the projector arena, the target was automatically re-sized without any user intervention to 12"x18" which is 1.5 times bigger .  This happens both programatically and when adding a new target to the projector arena using the shootOff main window.  This is causing me to re-scale my targets to the size I originally made in photoshop when it should not be necessary.

cbdmaul, 2016-05-27:
What are those measurements in pixels?

ifly53e, 2016-05-30:
72 pixels per inch

cbdmaul, 2016-05-31:
1. 72 pixels per inch in what? Photoshop?
2. What's the resolution of the screen the projector arena is on? What is its DPI set to? If different, what are the resolution and DPI of your primary display?

ifly53e, 2016-05-31:
1.  72 pixels per inch in what? Photoshop?  Yes.
2.  What's the resolution of the screen the projector arena is on? 1900x1056
 What is its DPI set to? I don't know...is that set by javafx?
 If different, what are the resolution and DPI of your primary display? 1900x1080 (not sure about dpi can I check that in windows somehow?)

cbdmaul, 2016-05-31:
1900x1056? Is that the size of the window? That's an odd resolution for a projector.

Checking the DPI depends on the version of Windows.  Try going to "run" or hitting Win+R and typing "dpiscaling".  There is a slider that indicates the current DPI scaling for the primary display.  On Windows 8 and newer, this setting is per display and you will need to go back to "Screen resolution", select a different display, and click "Make items bigger or smaller" to see each display DPI.

Note that on Windows, the default and lowest DPI is 96, not 72.  "Smaller" on the DPI slider is 96, "Medium" is 120, "Larger" is 144, and "Extra Large" is 192.

cbdmaul, 2016-05-31:
Also note, javafx does odd scaling of resolutions based on a pagan ritual involving pig's blood and a sacrificial goat.  We have multiple bugs open, #396 and #438, due to javafx's oddities.  It's possible even if we figure out how to correct for your situation that it won't be implementable for everyone until javafx is fixed.

cbdmaul, 2016-06-10:
If you don't have a followup on this, I am going to close this.  We are working on #424 which should make this possible without doing it manually.

ifly53e, 2016-06-10:
I am out of pig's blood and I am saving my goat for dynamically loading courses...keeping my fingers crossed...

closed 2016-06-10 by cbdmaul