phrack/ShootOFF #454: Shot's created in ShotDetectionManager hardcoded to zero time

ifly53e, 2016-06-26:
Request change from:
final Shot shot = new Shot(color.get(), x, y, 0, cameraManager.getFrameCount(), config.getMarkerRadius());

to:
final Shot shot = new Shot(color.get(), x, y, System.currentTimeMillis(), cameraManager.getFrameCount(), config.getMarkerRadius());

phrack, 2016-06-26:
The timestamp is actually set using the canvas manager later.

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/CanvasManager.java#L519

There is probably a cleaner way to do this because I don't remember why two shots are created but it does seem weird.

ifly53e, 2016-06-26:
I forgot about that one.  I ended up changing that time as well to just System.currentTimeMillis() instead of the difference that was originally there so I could see consistency in the differences between times that were returned to the functions relative to the system time.

Is there other code that uses the shot time for something?

Instead of time, I was considering using the shot's frame for the lag solution but I have not found a need to go there yet.

phrack, 2016-06-27:
It's used in the shot timer to show splits. It might also be used in the session recorder, but probably not. I am pretty sure the session recorder keeps track of time itself to play back sessions in real time.

commit c7f53b00 (2016-07-04): Consolidate shot creation operations to close #454
closed 2016-07-04 by phrack