phrack/ShootOFF #455: Target Interface previous included method missing

ifly53e, 2016-06-27:
Please add:
public Group getTargetGroup();
back to target.java for access to public method:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/TargetView.java#L141

Thanks.

phrack, 2016-06-27:
Is there a specific case you have where you can't just cast the Target to a TargetView?

```
Target t = ...

...

if (t instanceof TargetView == false) throw new AssertionException("TargetView is no longer in the type chain for Target, new version of ShootOFF?");

Group targetGroup = ((TargetView) t).getTargetGroup();
```

ifly53e, 2016-06-27:
Phrack,

This is what my code looks like with the requested change.

targetLeft.getTargetGroup().setRotate(rotation);

How do I get the same result without it?

Thanks.

phrack, 2016-06-27:
If you translate what I said to what you just wrote, it'd be:

```
if (!(targetLeft instanceof TargetView)) throw new AssertionError("TargetView is no longer in the type chain for Target, new version of ShootOFF?");

((TargetView) targetLeft).getTargetGroup().setRotate(rotation);
```

If you're trying to make some other point I don't get it. If you're not trying to make some other point, I wish you'd make more of an effort to read and understand what I wrote before running off assuming I am trying to be dismissive/arrogant/jerky/whatever.

ifly53e, 2016-06-28:
Listen, Mike.  I am an ok hacker and an average programmer.  When I ask a question, I am looking for guidance.  I have used some API's before.  Those API's were a lot easier to use.  You designed a framework that is not intuitive to me.  I am willing to bet, that my programming skills are slightly above the average user that will bump into your API.  If I have a question, I think it will be safe to assume that more than a few others will have the same one.  I wish you would think of my questions as opportunities to clarify the interface instead of attacks on your code.

My goal is to get my exercise designed according to a shooting test I am trying to pass.  That test requires scalable, timed, and rotating targets along with recoiled shot detection.  I have designed the exercise.  I am waiting for native shot detection to be implemented and lag to be fixed.  I am trying to help with the lag and offer my feedback on the API.  I am going to attack recoil next.

I think our goals are aligned.  I am assuming that you want a functional API that an average programmer can use?  Me too.  I read, re-read, and re-read your answers here and I spend hours trying to understand your design logic.  I think at this point, it is safe to assume that without a book on how the ShootOFF API was designed with explanations of how the user of the API is supposed to think about an exercise, it is not going to get any easier for me.  I read and re-read your wiki...multiple times.  

All that said, you should know that I have designed a nice set of exercises with your API that is helping my shooting.  Bear with me and my questions and I think that the end result will be a nice free tool to help engineers reverse their way to a low cost dry fire solution.

Tim

phrack, 2016-07-02:
I don't consider questions attacks. I consider comments like this to be attacks, which is what I was commenting on in this issue instead of the linked to one (I've unsubbed from that one for my own patience):

https://github.com/phrack/ShootOFF/issues/416#issuecomment-228548731

I have never levied an intential insult on you nor have I personally felt superior to anyone working on ShootOFF in some way. I feel that at worst I've answered questions in a terse manner, especially in that thread where I legitimately don't know what the right answer to the overall problem is.

In this particular case, it's not a framework issue. It's just Java. Eclipse has very nice type searching features that will tell you exactly where Target is implemented (it has to be implemented as a feature). It is only implemented in one place. You can therefore cast anything that's a Target to the one implenentation of Target (TargetView). 

I don't feel like I should have to teach people the language or the tools. If they don't know them and are unwilling to learn, I just can't be bothered to care.

Unsubbing from this issue as well.

closed 2016-07-02 by phrack