phrack/ShootOFF #458: If possible, resize anchors should stay a consistent and sane size

phrack, 2016-07-05:


cbdmaul, 2016-07-06:
Still seeing this if I make perspective changes?

[image]

phrack, 2016-07-06:
OK, I only had half the solution. I forgot that there are actually two places targets are scaled and my new tag is only respected in one of them. `TargetView.setDimensions` is the remaining culprit.

phrack, 2016-07-06:
It's also probably a bug that right clicking a target selects it for resizing purposes. That doesn't happen in most drawing programs for example.

phrack, 2016-07-07:
There are actually two problems here:

1. If you call setDimensions it doesn't respect the resizable = false tag. I tried doing the same thing that works for resizing with the mouse and for some reason it isn't work for me. I am probably tired and missing something small so I'll try again with a fresh slate some other time since this isn't super important (the dragging with the mouse case is far more important and that is fixed).

2. If you resize a target, unselect it, then reselect it the anchors get shrunkor expanded based on the target groups overall scale. Trying to scale it the other way to keep a consistent size (this is what the mouse solution, which works, does) is not working for some reason. For some reason the scale is being ignored when I am adding the anchors.

commit eb19c460 (2016-07-05): Added tag for unresizable regions and made resize anchors unresizable to close #458
commit 084af46e (2016-07-09): Keep resize anchors the same visual size for scaled targets to close #458
closed 2016-07-09 by phrack