phrack/ShootOFF #463: Race condition in processFrame

cbdmaul, 2016-07-07:
Exception in thread "ShotDetector" java.util.NoSuchElementException: No value present
	at java.util.Optional.get(Optional.java:135)
	at com.shootoff.camera.CameraManager.processFrame(CameraManager.java:551)
	at com.shootoff.camera.CameraManager.streamCameraFrames(CameraManager.java:485)
	at com.shootoff.camera.CameraManager.access$0(CameraManager.java:462)
	at com.shootoff.camera.CameraManager$VideoStreamer.run(CameraManager.java:457)
	at java.lang.Thread.run(Thread.java:745)

phrack, 2016-07-07:
Wow, that looks like one in a billion. You must have been recalibrating or closed the arena?

cbdmaul, 2016-07-07:
recalibrating

cbdmaul, 2016-07-08:
I have had this more than once btw.  But probably would never happen in regular user use.

phrack, 2016-07-09:
I can try it later, but it seems plausible to me that locking on projectionBounds isn't a big deal both because the critical sections would be small and because very little code touches it anyway. In other words, in this case locking shouldn't kill performance

commit 18b7a901 (2016-07-09): Fixed race condition on projectionBounds to close #463
closed 2016-07-09 by phrack