phrack/ShootOFF #469: Bias shot color detection more towards red.

phrack, 2016-07-08:
Green lasers are always brighter than red lasers but red lasers are far more common. Right now red is fairly common misdetected as green. We should shift the bias in detecting colors a bit more towards red's favor.

phrack, 2016-08-09:
This was fine at my house with red. Need @cbdmaul to test with green laser pointer and close if good enough.

cbdmaul, 2016-08-10:
I'm happy with how this is.  I was still able to get red shots detected as green but only in conditions where the shot looked damn green to me in the video.
[image]

phrack, 2016-08-10:
Was green ever detected as red?

cbdmaul, 2016-08-10:
Yes.  From looking at our color metrics in [WEKA 3](http://www.cs.waikato.ac.nz/ml/weka/) I can tell they're not good enough to accurately distinguish red from green regardless of how we bias it.  Better metrics might help but that's about it.

phrack, 2016-08-10:
I don't think anything will make a big difference beyond adding a color mixing model that can predict what red or green would do to the color that's already there (e.g. sky + red = approximately this green, sky + green = approximately this green). 

I guess my real question is: does it detect green on targets as green enough to close this issue?

cbdmaul, 2016-08-10:
It still gets green right the majority of the time.  All the tests pass just the same as before with some very slight improvements to picking up red.  I could bias it further to red but I'm not sure that will actually improve anything, like I said the bias is not the only issue, the accuracy of color recognition is the bigger issue here.

phrack, 2016-08-10:
OK, I am going to close this then. We can open an issue to add a color mixing model (or something else) if we ever decide this is super important. For now, this biasing is good enough IMO.

closed 2016-08-10 by phrack