phrack/ShootOFF #472: Translate clicks in the feed window into clicks on the projector arena when autocalibrated

cbdmaul, 2016-07-09:
1. Autocalibration required for very good alignment (otherwise, would be much too confusing to users, I would rather it didn't work at all than have unexpected behavior)

2. Check the autocalibrated bounds area of the camera feed when there are feed clicks

3. If the click was not on a shot marker, pass them to the arena (In the autocalibated bounds)

4. The arena will have to translate them to the size of the projector resolution

5. Handle them like a normal click in the arena

It will likely require disabling shot detection when any target is selected or moved, so a warning to the user may be necessary.

I'll work on this issue today if my other duties go smoothly.  I want it bad enough.

phrack, 2016-07-16:
6. If a target was added to the feed (using Target -> Add Target) over the calibrated region, the feed target gets priority over translating the mouse movements. It would be confusing to move a target on the feed and have one on the arena move at the same time simply because it was under the target you intended to move. This may sound obvious but JavaFX is permiscuous about passing events to everyone within a group or that has a parent/child relationship.

It might also be useful to "highlight" the calibrated region in yellow on the feed when the cursor is over it. This would help you see that your cursor is in the right place while giving you simulated arena bounds without having to focus on the feed too closely. Probably not worth even trying the highlighting idea unless doing the rest suggests it's necessary.

phrack, 2016-07-16:
Follow-up on highlight idea: it might be a useful signal to the user that they _can_ move targets on the arena from the feed if they so desire. See the highlight = can move, don't see a highlight = it won't work.

phrack, 2016-08-13:
I actually forgot I removed the button code. In CanvasManager replace

```java
	private void translateConvertedEventToTarget(MouseEvent event) {
```

with 

```java
	private void translateConvertedEventToTarget(MouseEvent event) {
		Button x = new Button("XXXXXX");
		x.setLayoutX(event.getX());
		x.setLayoutY(event.getY());
		canvasGroup.getChildren().add(x);
```

cbdmaul, 2016-08-15:
This works great for moving targets around now...but resizing seems to go nuts.  Presumably need to recur down from the group level to the node and adjust the event coords again... if you agree...I can think about what that would take to code.

phrack, 2016-08-15:
Yes, I think that's correct. Resizing should be based on dragging at the node level (an anchor node).

cbdmaul, 2016-08-17:
The problem is the eccentric design of the targetview's handling of events.  It is nothing like the examples I found online and I have trouble even understanding how the hell it works right.

For example, x and y are stored on onmousemoved, then used to determine deltas for moving.  But resizing doesn't use that, despite being in the same onmousedragged handler.  How the fuck does resizing work?

As a side node, onmousemoved is not triggered in CanvasManager.  I don't know why.  I'd like to know why.  The handler is registered.

I think we might need to bring targetview to a reasonable design (for example but not necessarily the best: https://java-buddy.blogspot.com/2013/07/javafx-drag-and-move-something.html) before coming back to this issue

phrack, 2016-08-17:
If there are actually enough examples that are really our case and not just loosely kinda our case for the current working solution (except pass through) to be eccentric, I'll be pleasantly surprised. I'll be ecstatic if there is a simplification that genuinely simplifies the feature set. Here is the example I started from, it was basically all that was out there at the time this code was first written that was even close: http://andrewtill.blogspot.com/2012/12/dragging-to-resize-javafx-region.html. 

Requirements:

- All regions in a target must stay the same size relative to each other when the target is resized.
- All regions in a target must stay the same position relative to each other when they target is resized.
- All regions in a target must stay the same size and position relative to each other when animated (except for those changes expected due to the animation). 
- All parts of a region that a user can see that would be shootable in a real target must be shootable in the virtual target.

All of the examples last time I looked were part of just one of these. They all resized or moved one thing at a time and indepently of every other node. They all also demonstrated themselves on a [Shape](https://docs.oracle.com/javase/8/javafx/api/javafx/scene/shape/Shape.html) or some other JavaFX specialization of a Node that comes with primitives to make resizing and moving much easier than what you get from [Node](https://docs.oracle.com/javase/8/javafx/api/javafx/scene/Node.html). We have two complications:

- We have to not only resize one thing, but many things all at the same time. The only targets we have that are just one region are the silhouettes, one of the poppers, and the reset target. Only the reset target uses a Shape. The regions must stay the same size relative to each other or the target breaks. Their origins won't move if you resize each individually without a position adjustment so the target will also break if that isn't handled. The position adjustment is hard to get right if you do decide to resize individually. One of the JavaFX's transformation classes has a hint at the formula, but I haven't found it again this morning. I found it's best to just do it at the group level so JavaFX will deal with it.
- We have Shapes and [ImageViews](https://docs.oracle.com/javase/8/javafx/api/javafx/scene/image/ImageView.html) in many of our targets. The only common type for these is Node, which does not have specialized convenience methods.

Having established that, here are complications I found:

- All Nodes reset the coordinate system internally.
- All Nodes provide three different views of their coordinate system, only the local one and the parent one are useful I think.
- All Nodes can have transformations applied to them. These transformations are affected by any transformations applied to their parents, so that has to be accounted for.
- It's easy to resize a region visually and not realize it was not "physically" resized. What I mean by that is the target will look the right size but the exposed views of the coordinates are not transformed so unless you transform them yourself to detect a hit or physically resize the target, you will miss actual hits.

Regarding the x and y fields. Those were used for moving and resizing with the mouse because I started from the example I linked to which uses them. In fixing resizing a few versions ago I got rid of them for resizing, but not for moving (I haven't touched the moving code since the first version because it always worked). They can probably be removed for moving too.

cbdmaul, 2016-08-17:
Can you address the onmousemoved issue?  It's unclear that we'd need it in canvasmanager but I'd like to know why it isn't working.

The problem right now is that TargetView abuses the mouseevent coordinate systems.  It uses them differently for resize and move, and relies heavily on onmousemoved.  If move worked like resize, I think I could come up with a uniform solution on the canvasmanager side.

phrack, 2016-08-17:
I don't know how it's abusing them beyond shoving a target model into a GUI model (I don't know how else to create the target model given the tools available). I don't see a blatantly right way to do this since we have a unique and complicated set of requirements. 

The only difference between the two approaches is one calculates the delta for moving using the last mouse coords and the other calculates the resize delta using the targets current origin. We should probably just remove the x and y fields and let move us the current target origin too since this has less global state. I can't give input on the mouse move issue without being able to navigate the code easier than I can on a phone.

phrack, 2016-08-17:
I should also clarify for historical reasons that the move code always worked based on the blog example and was never touched again because people generally don't change code that is small and meets the requirements. The resize code didn't work which is why it evolved away from the design moving uses. It makes sense to make them use the same model to meet our new requirement but it's unreasonable to expect the code to be perfect for a new requirement right away. Particularly when the original set of requirements was already challenging given the available tools.

cbdmaul, 2016-08-17:
The move code does not work like the blog shows, mouse moved is not used for coordinates in the blog example.

Once again I can't comply with what TargetView currently expects unless I generated fake mouse move events because canvasmanager is not receiving them.  Plus using mouse moved for dragging events is not necessary anyway.

I think it is an abuse when two identical inputs have different processing based on state (move or resize) that is only accessible to the internal class.  What I send from CanvasManager to TargetView should either work for both drag and resize, or work for neither.

phrack, 2016-08-17:
The canvas isn't getting the mouse move events because the background is. For whatever reason, JavaFX chooses to address that particular mouse event and not the others to the background. 

`canvas.setOnMouseMoved(passThroughHandler);`

should be:

`background.setOnMouseMoved(passThroughHandler);`

There is documentation on this that is slightly clearer than mud here:

https://docs.oracle.com/javase/8/javafx/api/javafx/event/EventDispatchChain.html

phrack, 2016-08-17:
This feature has a ton of user experience issues.

The resize anchors are always false shot city with pass through, but especially when a target is moved (which is buggy, it jumps around in weird ways). I imagine this could be an issue with moving a target around even if resize anchors are not present. This leaves little option but to disable shot detection when a target on the arena is being manipulated with the mouse in any way. You can't solve it with a visible light filter or lower exposure because then you couldn't even see the target in the feed to try to move it. 

We of course would need to notify the user of shot detection being disabled, otherwise people will have no idea why their shots aren't being detected. The canvas group that is passing events through has no idea if you are manipulating a target because it has no idea what happens with the events on the other end. The target knows it is being manipulated and it knows what canvas it belongs to, which can be used to get the camera manager to disable shot detection (gross, but it works). However, the canvas doesn't know it belongs to the arena so there isn't really a way to only disable for targets on the arena instead of anywhere. Even worse, the canvas where the target is doesn't know what canvas is passing the messages through so there isn't a good way to know where to even show the diagnostic message.

Furthermore, you don't get any feedback about whether or not you are on a resize area. Is it far enough over to resize or will it move the target? Try it and find out! Passing messages through to nodes that they weren't intended for is already a hack, but passing feedback the other way is going to be ugly as sin. 

I am just not sure that this feature is really worth it. I can't really speak to all projectors, but with mine I just watch the laptop screen while I move and resize targets on the arena because I can see it all just fine in the camera feed. Is a tiny bit of convenience when using a visible light laser on an unconfigured camera worth solving a large number of non-trivial user experience issues for? It wouldn't even be a problem with tablets becase we'd always be showing you a view of the arena that will never lead to a cursor or resize anchors. There would be no cursor and we can just show the resize anchors on the tablet and send a message to the backend saying, "OK, now this target should be this big."

phrack, 2016-08-17:
To hopefully finish the killing of this issue, I propose an alternative:

In 3.9 when we are making the GUI easier to use, let's add a tab with the cameras for the arena. This tab will show an exact duplicate of what is being projected. In fact, it may even be the exact same instance of the canvas group on the arena we are projecting so that anything you do on one is instantly mirrored on the other. This has the following benefits:

1. No more having to think about how the users puts targets on the arena instead of a camera feed. They always just click one button to add a target and the target gets added to the canvas for whatever tab is selected.

2. Never a need to move the mouse to the arena. Even with a visible light filter (or not) you'd just use the arena tab because it's easy.

3. Way easier to implement and mirrors the user experience I expect on tablets (where you aren't really working directly with what's projected -- although you could still have that option on PC since it's mirrored).

phrack, 2016-08-17:
This issue has been superseded by #507.

closed 2016-08-17 by phrack