phrack/ShootOFF #476: Undeterministic or failing auto calibration test results?

cbdmaul, 2016-07-10:
Our build failed last night, but those tests are passing for me.

I did a project update, and one test failed the first time...but worked after that.  I'm wondering if there is something undeterministic here.

phrack, 2016-07-10:
This is pretty consistent in travis for some reason.

cbdmaul, 2016-07-10:
Do they work for you?

phrack, 2016-07-11:
These autocalibration tests are fairly consistently failing for me as are the perspective tests now failing in travis too. For the latter, I think you revised some calculations that changed results and the perspective tests weren't updated to account for that.

phrack, 2016-07-11:
PS. The autocalibration ones are failing because pixels (usually early on) don't match. For example, for testCalibrateProjection it fails at 0, 0 because imgA is -12640756 and imgB is -12574702. Maybe a test resource isn't updated in the repo?

cbdmaul, 2016-07-11:
My copy is up to date and everything is committed.  Tests pass.

cbdmaul, 2016-07-15:
We still have this problem, possibly due to architecture differences in opencv.

phrack, 2016-07-17:
Fixed for me, suppose we can wait to close this until Travis agrees.

cbdmaul, 2016-07-17:
Yeah, I was waiting for it but I've got to go.

phrack, 2016-07-17:
I am going hiking today but the test should finish before I leave. I'll close it.

commit 6d3992e7 (2016-07-11): Closes #476
closed 2016-07-17 by phrack