phrack/ShootOFF #481: Paper pattern is not discovered when full pattern causes camera exposure to change

cbdmaul, 2016-07-10:


cbdmaul, 2016-07-24:
I have committed something that works with tests but I have no projector to test with at my current location.

phrack, 2016-07-24:
I can try to fix this. So far, aside from this exception, it has been working very well.

Exception in thread "ShotDetector" java.util.NoSuchElementException: No value present
	at java.util.Optional.get(Optional.java:135)
	at com.shootoff.camera.autocalibration.AutoCalibrationManager.findPatterns(AutoCalibrationManager.java:344)
	at com.shootoff.camera.autocalibration.AutoCalibrationManager.stepOne(AutoCalibrationManager.java:303)
	at com.shootoff.camera.autocalibration.AutoCalibrationManager.processFrame(AutoCalibrationManager.java:214)

phrack, 2016-07-24:
I'll actually have to leave that exception to you because my attempts to fix it (just adding a break or a continue if it wasn't present) seemed to hopeless break things.

It was working in conditions like this:

[image]

This seemed to be the first breaking point, but I don't think it's even fair since you can't really see the pattern in any condition:

[image]

In addition to that exception, which became quite frequent the more I tested, the most common problem was that the frame would get stuck in some intermediate state when it would try to find the pattern, then obviously it never found the pattern. Here is an example where the message telling you that you need to calibrate never quite went away:

[image]

I got a bunch of different variations of this including the arena not being all the way up yet, the arena being up but the pattern not being up yet, etc.

phrack, 2016-07-24:
I am also getting mask-0..3.png dumped to the file system.

cbdmaul, 2016-07-24:
I know what the problems are, working on a fix.

phrack, 2016-07-24:
I knew that would be the case, thus I still have everything set up for when you give me a thumbs up to test again.

cbdmaul, 2016-07-24:
Should be fixed now.  Your paper pattern is upside down (which doesn't matter, we just don't have a test for that).

As for the ACM starting too quickly, seems like that should be on the manager's side and not acm's problem

phrack, 2016-07-24:
I think the ACM starting too quickly issue was actually due to the exception and my non-solutions to it. I can't dupe it anymore even in the same lighting conditions. In fact, it's working very well now even in conditions it didn't before.

closed 2016-07-24 by phrack

phrack, 2016-07-24:
PS. I think it's safe to say there is no upside down because it's been working fine for me all day that way. Working on the CFDA exercise.

cbdmaul, 2016-07-24:
I added logic to sort the corners

commit a2b6470e (2016-07-24): Run autoformatter to close #481