phrack/ShootOFF #486: Plugin URL is hardcoded

ifly53e, 2016-07-27:
Currently the URL for plugins is hard coded to:
http://shootoffapp.com/shootoff-plugins.xml
and a pull request is required to change that file to include additional plugins.

Request the ability for the user to input additional URL's that host properly formatted plugins.xml files in order to allow other entities to host their own exercises.

phrack, 2016-07-27:
Do you have something you wanted added to the list for now? I'll put basically anything in the plugin list since the code doesn't live in a repository I have to support.

ifly53e, 2016-07-28:
I don't have exercises to add for now. 

Allowing URL's to a developers own plugins.xml file would also unburden you from the responsibility of updating the shootOFF plugins.xml file every time a developer requires an update to a name, version, download location, or description.

I am guessing that changes to the pluginManagerController would be required to add additional plugins from a persistent and editable list of URL's would be required.

Do you think that adding to the list would be better implemented for the user from the Exercise Manager or the User Preferences?

phrack, 2016-07-29:
I don't think the implementation is hard in this case, the user experience is. You're basically proposing the equivalent to Eclipse update sites and that user experience is iffy even for developers. I am not sure what I'd present to audiences that find even our current GUI difficult (hence the need to make it more usable).