phrack/ShootOFF #489: Warn users when hardware is insufficient for good performance

phrack, 2016-08-08:


phrack, 2016-08-09:
@cbdmaul The PassMark data does not have an API. Dumps of the data start at $500. There is an unofficial API, but I am not sure how kosher it is or how reliable it is.

https://github.com/yakka34/Unofficial-cpubenchmark.net-API

cbdmaul, 2016-08-09:
That API looks decent.  Worst thing that happens is that we have to fall back on a different technique, like total memory.  We will probably need to add it to the startup warning dialog and ask the user to agree to it.

I'm assuming getting the CPU name isn't too hard in java...

phrack, 2016-08-09:
You need to use native code to get the CPU name, but this is a solved problem because ShootOFF's diagnostic tool already does precisely this using a cross-platform library. 

Seems to make sense to try to get PassMark score and warn on first run if the score is <4000. Fallback to warning if the system has less than 4 GB of RAM.

phrack, 2016-08-09:
PS. The warning could be part of the disclaimer. If your system is too slow you get big read text saying "poor performance likely" or something with a quick explanation.

phrack, 2016-08-12:
ce52815b3c466c33579842c2e843da39333ab60b

closed 2016-08-12 by phrack