phrack/ShootOFF #490: Attempt to lower FPS of webcam feed for calibrated camera to improve free up cycles for other operations

phrack, 2016-08-08:
Only do this when arena is open and camera is calibrated.

phrack, 2016-08-09:
@cbdmaul Profiled in a number of different conditions. With default arena settings the code to display frames is in 7th place for the mostly costly operation, but it weighs in at a mere .1% of execution time. The first six are shot detection operations. With hasher settings (e.g. cropping and lots of skew) this crawls up to about .7% of execution time, but it's still in 7th place. The shot detection code is just that intense.

That said, can still play with it  a bit to see what happens to the effective FPS, but the profiling numbers suggest no miracles will happen.

phrack, 2016-08-09:
OK. I did the following tests with the following results:

- Measured FPS with bright and dark feeds showing all frames (current case) vs. only ever showing the very first frame then never showing a frame again (best possible optimization). **FPS Result:** No change. This suggests, as the profiling data also supports, that showing the webcam feed isn't intensive enough to make a difference.

- Measured FPS with bright and dark feeds making bright pixels red (current case) vs. not providing the diagnostic information while always showing the frames (current case with best optimization for diagnostic drawing). **FPS Result:** not drawing bright pixels consistently achieved a 1-2 FPS improvement for very bright feeds (I was making essentially the entire feed so bright it'd get drawn red) but it was not meaningful for typical bright feeds I get that will cause a brightness warning. 

My conclusions:

1. It's not worth dropping the frame rate in any case for the webcam feed.
2. For recommended machines throttling the drawing of diagnostic pixels is also not worth it, but on lower end machines it could eek out just enough additional FPS in bright cases to make shots in non-bright areas more detectable. Having said that, in these cases most of the feed isn't shootable anyway because we just won't detect shots on most of it.

My overall assessment is nice idea, but we should close this issue without doing anything about it. I'll wait for @cbdmaul to weigh in first though.

cbdmaul, 2016-08-09:
I think it will be a more significant change for higher fps / native shot detection situations, I am going to test the 120fps cam today, I'll give this a quick check.

phrack, 2016-08-10:
I was very close to maxing out my FPS for the non-bright tests. It's possible high FPS will make these kinds of small optimizations more important, but my guess is that shot detection will also just increase in intensity to such a degree as to drown out these operations.

cbdmaul, 2016-08-10:
Did your testing include not converting back to bufferedimage for frames that don't need it? I'm wondering if that is more intensive than displaying them.

My own testing was inconclusive, FPS was kind of all over the place.  High FPS camera never did more than 37-38 FPS so it's hard to tell.

phrack, 2016-08-10:
I forgot to remove that, but I don't expect it to change the results much if any. The profiler data ranks that code even below the code that displays the frames (so it's less than .1% of the execution).

Code that updates shot detection filters and searches for shots absolutely dominates execution time according to the profiler. Anything that isn't directly related to finding a shot is a drop in the ocean according to all of the data I've seen this year.

phrack, 2016-08-10:
PS. According to the profiling data the two biggest ways to have an impact would be:

1. Do some deep thinking about if there is a way to do no color conversions. If we could read all of the frames off in BGR and display them that way it'd be huge.

2. Don't update the filters for every pixel on every frame (I've played with variations of this in the past and could usually get 25% speed improvements but at huge accuracy penalties).

phrack, 2016-08-10:
Was thinking about this more at the gym. Eliminating pixel format conversions and reducing filter updates would be big, but they're also hard and have seen attention already.

Here is what I really think we should do: close this issue because I don't think we're going to get serious gains out of it, but open an issue to try to replace getting frames from the camera with OpenCV. If this is a big win it will be because webcam capture's webcam lock is always in our top 6 (towards the top during periods with no shots). Assuming the indexes for cameras in OpenCV are the same as the ones in webcam-capture (I don't see why not, would have to test it), we would still use webcam-capture for configuration but OpenCV for the rest. If the indexes match this could potentially get 20%+ improvement without doing more than an hour or so of work.

cbdmaul, 2016-08-10:
I agree

phrack, 2016-08-10:
If you dupe my OpenCV hack results (email coming shortly), we might want to re-open this. Displaying the frames goes from a mere .1% of execution to 10% because OpenCV gets rid of a lot of horrible overhead in shot detection.

phrack, 2016-08-11:
Bringing this back to life because #491 reduces shot detection overhead enough that toFXImage to display the feed goes up to 10% of execution. Given what we've learned between this and 491, I suggest we close this with a super simple chane to CanvasManager.updateBackground that artificially limits the FPS all feeds to 15 FPS. That's fast enough to be usable and look fine IMO while also ensuring this will never be a severe source of overhead.

commit 41612bc2 (2016-08-12): Limit webcam feed FPS to close #490
closed 2016-08-12 by phrack