phrack/ShootOFF #496: Exception thrown when enabling shot to click while calibrating

cbdmaul, 2016-08-14:
Yes...I know...don't do that

Note line numbers in canvas manager probably don't match what you have, but this is easily reproducible so we can handle it later.

Exception in thread "JavaFX Application Thread" java.lang.AbstractMethodError: com.shootoff.plugins.ShotToClick.targetUpdate(Lcom/shootoff/targets/Target;Lcom/shootoff/plugins/TrainingExercise$TargetChange;)V
	at com.shootoff.gui.CanvasManager.addTarget(CanvasManager.java:868)
	at com.shootoff.gui.CanvasManager.addTarget(CanvasManager.java:851)
	at com.shootoff.gui.CalibrationManager.createCalibrationTarget(CalibrationManager.java:187)
	at com.shootoff.gui.CalibrationManager.enableManualCalibration(CalibrationManager.java:218)
	at com.shootoff.gui.CalibrationManager.lambda$4(CalibrationManager.java:299)
	at com.sun.javafx.application.PlatformImpl.lambda$null$173(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)

phrack, 2016-08-14:
Huh, that implies it's an old version of the exercise from before the target update. Where did you get this exercise from?

cbdmaul, 2016-08-14:
Ah, I see, it is from April.  I didn't even think about what version it was.  Am I supposed to be able to update it somehow? It's not in the get exercises menu.

phrack, 2016-08-14:
In this case I recommend just deleting the shot to click jar in your exercises directory manually and try again in the exercise manager.

cbdmaul, 2016-08-14:
No exception with latest shot to click.

closed 2016-08-14 by cbdmaul