phrack/ShootOFF #498: Add button to open camera settings dialog

cbdmaul, 2016-08-14:
Now that we are using opencv, we can invoke the camera's native settings dialog with:

camera.set(CV_CAP_PROP_SETTINGS, 1);

It would be very useful to have a button for this somewhere.

phrack, 2016-08-14:
Not too surprising, but this doesn't work on Linux and I doubt it does on Mac either -- you get a "property unsupported" error. Assuming it does in fact work on Windows, I suggest adding it to the right click menu for camera feeds now via a "Configure Camera" button. The code to add this button should only execute if the OS is Windows (check the Main.main to see how to do that).

phrack, 2016-08-14:
PS. This should be so quick and easy to do I suggest just doing it in 3.8 if the feature works on Windows.

commit cda44503 (2016-08-14): Closes #498
closed 2016-08-14 by cbdmaul