phrack/ShootOFF #507: Use a tab located with the camera tabs to control the arena

phrack, 2016-08-17:
This issue replaces the pass through idea from #472. See this comment for details: https://github.com/phrack/ShootOFF/issues/472#issuecomment-240575465.

cbdmaul, 2016-08-17:
Does anybody use the paper targets at the same time as the projector targets? Can we just have the projector targets menu be the normal targets menu on a cameras with arena?  This plus defaulting to the arena view would be a huge usability improvement in my opinion.

Although I wouldn't suggest it now it is likely possible to automatically detect in some cases when the alignment between the camera and projector is wrong

phrack, 2016-08-18:
I did once because I was using so many virtual targets that I wanted my reset target to be away from my projector real estate. I did it another time again because I wanted a reset target that would not get in the way of targets moving around the screen. I did it a third time because I was switching courses a lot and wanted a reset target in a static place that, again, would never get in the way of what was on the projection.

I am not sure how common that use case is or if it ever extends beyond wanting a reset target that is never in the way of what's happening on the arena. If it is only ever the reset case I used we could maybe do something else, like a shot off the arena is (configurably) a reset.

This is another kind of thing that is easier on a tablet because we could use VTT to say, "Reset" instead of needing to shoot something.

cbdmaul, 2016-08-30:
1. arena stage in main window is full sized, not the size of the space available
2. moving target to top corner in main stage causes the background to stretch out and target does not move in projection stage
3. moved target in projection arena stage to bottom right corner and the target disappeared from the main window
4. manual calibration is very confusing now because you have to move to the other pane to do it.

phrack, 2016-08-30:
For 1, it could be scaled relative easily since targets just need to be scaled in MirroredTargets and shots  before the call to mirrorAddShot. Having said that, I suspect both copies need to stay the same aspect ratio or it will be annoying. If that's true, it would likely never really fit right unless the tab expands as the main window grows to not need scroll bars.

Is 4 as easy as switching to the calibrating camera feed when auto fails?

cbdmaul, 2016-08-30:
1. Keep the aspect ratio the same and make one edge the longest possible without the other exceeding the available space.  For instance, set newWidth = window stage width, newHeight = newWidth * Aspect Ratio.  If newHeight > window stage width, then do the reverse.  newHeight = window stage height, newWidth = newHeight * Aspect Ratio.  It has to fit one of those two ways

4. Yes, I think so.  When we get to design, we can highlight the "Stop calibration" button and/or even put instructions in the bottom tray.  For now let's just get them to the right place.

I don't like manual calibration at all, I could see deprecating it to testing only in the near future.

phrack, 2016-08-30:
The numbering isn't consistent in the GitHub email vs on the website.

In any case, I fixed a previously unknown bug with resource targets (e.g. Clay Busters) and did the tab switching on manual-calibration. I tried to mitigate the background warping issue by using TargetView.setKeepInBound (previously used for just the manual calibration target). This didn't really work so I'll have to look again.

I don't even want to think about have scale mismatches until the real kinks are worked out and perspective projection is integrated in some better way than the original popup.

commit 494e7699 (2016-09-03): Fix click-to-shoot for arena tab to close #507
closed 2016-09-03 by phrack