phrack/ShootOFF #508: Make ShootOFF maximize by default and turn top level menus into large buttons

phrack, 2016-08-18:
https://drive.google.com/open?id=0B5SE-FRATIorMXJPTmhJdjhVd0U

Yellow buttons place a full window pane over maximized screen to show file options, target selection, settings, etc.

cbdmaul, 2016-08-30:
Here's a good design guide: https://msdn.microsoft.com/en-us/windows/uwp/style/index

Especially helpful are designs for controls that work on a variety of devices well: https://msdn.microsoft.com/en-us/windows/uwp/controls-and-patterns/index

cbdmaul, 2016-08-30:
This site has templates and examples of a lot of Metro/Modern design: https://metroui.org.ua/

phrack, 2016-09-05:
I know you're probably having a lot of fun with cameras but it'd be nice if we could make a slight trade for a bit. How does this sound? I'll fix bugs in existing GUI code if you take a shot at making things comply with design guides. This way you can try to make things look nicer but as soon as you run into a problem, you don't spend time on fixing it.

I am very tired of GUI code and it is not fun.

phrack, 2016-09-10:
@cbdmaul I did a little experiment with the slide controls where I ensured they always used up the maximum amount of space such that the buttons never grew and the padding was always 30 pixels on the edges of the control container. This means that the buttons are the same size but farther apart when the window is bigger.

We do not want to do that. Spreading the buttons out makes them harder to click because I have to move the mouse father and it breaks muscle memory if the window is a different size than what I am used to. It turns out that I always want them the same size and distance from each other and even centered in the window, otherwise my muscle memory doesn't take me to the right button anymore.

phrack, 2016-09-10:
IMO the main window is done when #544 is closed. The remaining problem for this issue will be the item selection pane because there number of button columns will no longe rmake sense for bigger or smaller versions of the window. This should be dynamically calculated.

phrack, 2016-09-11:
Anything that was possibly this issue that isn't done now should be opened in a new issue since it's already crowded in here. If anything is in that category, it's not actually maximizing by default. Even if we want that behavior outside of debug mode, we DO NOT want it in debug mode. Maximizing in debug mode means you have to go out of the way to end maximization to see log output. Seeing log output is a requirement essentially every time ShootOFF is opened in debug mode.

commit a1f7286e (2016-09-11): Adjust ItemSelectionPane column count based on window width to close #508
closed 2016-09-11 by phrack