phrack/ShootOFF #527: NPE with shot-to-click on arena tab

phrack, 2016-09-02:
This broke with pull request accepted as #524 

Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
	at com.shootoff.camera.CameraManager.getCurrentFrameTimestamp(CameraManager.java:728)
	at com.shootoff.camera.shotdetection.ShotDetector.addShot(ShotDetector.java:87)
	at com.shootoff.camera.CameraManager.injectShot(CameraManager.java:253)

phrack, 2016-09-02:
@cbdmaul 

This suggests either the timestamp method is in the wrong place or we need a null camera that doesn't actually get frames for the arena.

cbdmaul, 2016-09-03:
I fixed the NPE.  I noticed that a background has to be set for shot to click to work in the arena tab (Even when NPE was still in play).  Differ to you to close this and open new issue for that if necessary.

phrack, 2016-09-03:
I don't get the NPE anymore but it's still not showing a click-to-shot shot. I am not sure why but this is the last known issue for me for the arena tab so I'll keep looking. I haven't tested session recording yet so I am sure it gets worse.

In any case, I am OK with it not working on a synthetic shot on a blank background as long as it works on a target on a blank background. In other words, open the arena, let it calibrate, and shift+left click the background. Get nothing? OK. Add a target and shift+left click the target. You should get a shot.

Anyway, closing this because the remaining issue was captured in #507 last night.

closed 2016-09-03 by phrack