phrack/ShootOFF #535: Various bugs with prospective gui

cbdmaul, 2016-09-05:
1. Perspective panel does not go away if target is deleted

2. Changing the target width makes the target tiny.  Changing the height fixes it again.

phrack, 2016-09-05:
Regarding #2. The GUI doesn't appear to do anything that would mix up or return the wrong values. Aside from new validation code (to validate on the fly), almost all of the code is the same as it was before since it's visual changes. You could com.shootoff.gui.pane.TargetDistancePane in trace and see what this message says: "New target settings from distance settings pane: current width...".

If it all looks correct, it's definitely not the GUI.

phrack, 2016-09-05:
- [ ] Target deleted using mouse
- [ ] Target deleted using CanvasManage.clearTargets()

commit 4e178c2e (2016-09-05): Remove target distance settings when selected target is deleted to close #535
closed 2016-09-05 by phrack