phrack/ShootOFF #538: Request:  Add local file functionality for targets and courses

ifly53e, 2016-09-06:
Request the option to open local target files and local course files similar to the functionality found in getting a local background file.  Thanks.

phrack, 2016-09-06:
What's the use case? This is much harder for files and courses than backgrounds since backgrounds have no dependencies on external resources and targets and courses do.

ifly53e, 2016-09-07:
I have courses for both windows and linux because for whatever reason, the paths work in windows, but they do not work in linux.  It would be nice to be able to switch courses locally.  Also, I have several folders with different versions of ShootOFF with different functionalities.  Having a central local repository would save space on my thumb drive (which I am using to boot Linux from), instead of replicating the targets and courses in every ShootOFF instance.  

I hear what you are saying about the external resources but once they are set inside a target or course file via the unchanging path, doesn't the target file or course file just become a stepping off point for ShootOFF to find that external resource?  At that point we are just giving ShootOFF a local link to file that contains the information for where to open the external resource, right?

phrack, 2016-09-09:
Yea, I get what you're saying and generally agree. I'll leave this open and label it.

In any case, the path issue on Windows vs. Linux sounds like a bug. Can you open a bug issue for it and attach the contents of a course that has this problem? Just put the contents in a GitHub code block if nothing else.