phrack/ShootOFF #542: Calibrating with a camera, then disabling that one an selecting another, breaks projector mode

cbdmaul, 2016-09-07:
After you do it, the calibration button does nothing.

phrack, 2016-09-11:
Seems like something clean to do would be to add a `setOnClose` listener to cameras. Then we can cleanly add a close listener to the calibrating camera. If it closes, we clean up projector state to the point that this will work right. Might even be a good idea to just completely close the arena too. Then you have to click projector again to redo everything, but you also just completely changed your cameras so it doesn't seem like a big deal.

phrack, 2016-09-11:
I prototyped out the proposed solution and it worked well aside from the fact that the arena stage's on close request hander would get called twice every time. This is a problem because it tries to double up on clean up that causes exceptions. I have yet to figure out why the double call happens.

cbdmaul, 2016-09-13:
I see the same type of double/triple cleanup on the camera stuff.  If you find a good way to test it, we should deploy it on some of the closeable objects to try to weed out repetitive closures.

phrack, 2016-09-14:
In this case, I inserted the following code to try to determine why the `onCloseRequest` dupes happened using a stack trace:

`new Exception().printStackTrace();`

The stack traces suggest it's due to event bubbling and not some other location in ShootOFF causing the clean-up to happen again. My original code would blatantly cause this to happen based on my understanding of the JavaFX event system, but my current code shouldn't so obviously I am confused.

This is pushing me harder towards actually implementing the use of `Closeable` all the way:

1. Register a closeable when it's constructed.
2. Ask the registration class to close `Closeable`s, never do it directly. The registration system should atomically close and de-register so that closing can't be duped.

commit f1c35fbb (2016-09-16): Close arena when the camera that calibrated it closes to fix #542
closed 2016-09-16 by phrack