phrack/ShootOFF #546: Too much scrolling on the arena tab, either make it smaller or expand the viewport when the main window is bigger

phrack, 2016-09-09:


cbdmaul, 2016-09-13:
The feed/arena should expand to fill empty space.  The buttons at top can max out on height and fill space horizontally to expand to bigger controls.  See: https://metroui.org.ua/tiles.html especially the "Cloud Service" tile example.  This type of style could be used with the target/training buttons as well, especially with the mouse over showing the description.

Overall, things need to be bigger and brighter with individual controls easier to use with less sophisticated input (Either a novice computer user or a touch screen, similar operating conditions).  It is not necessarily the case, as you pointed out on the other issue, that we want things to just get huge when you full screen or have a larger screen.  Instead we want to have more info shown on larger screens, and on smaller ones the info will slide out instead.  For instance, the buttons at top might be minimal on smaller/windowed mode, but will slide down when the user hovers (Or drags down on touch screen).  Similarly for the bottom menu, so that the main feed window dominates on a smaller screen, unless the user specifically goes to the menu area.

Other examples:

https://metroui.org.ua/tables.html - Toggle switches that are more touchscreen friendly
https://metroui.org.ua/keypad.html - For allowing number input on touchscreens
https://metroui.org.ua/fluent-menu.html - This shows how a Windows ribbon works - makes it a bit clearer what submenu you are currently in, users won't get lost as easily
https://metroui.org.ua/tiles.html - Also on the tiles page, notice how mouse overs provide user guidance on what they're about to click on, and how the currently selected item has a big check on it, making it more obvious

We could also launch in a "hint mode" that uses hint overlays on all the buttons - https://metroui.org.ua/hints.html on the first run or until it is dismissed.

Here's a blog post about handling different resolutions for a common interface (I tried to find a better one but this will do for now): https://blogs.msdn.microsoft.com/b8/2012/03/21/scaling-to-different-screens/

cbdmaul, 2016-09-13:
To clarify a bit more, I think we should take the advice of the Microsoft blog and target no smaller than 1024x768.  To that end, let's design an interface that looks good at 1024x768 at a standard DPI (I think 72 dpi is standard?)  On a screen that is too small to show 1024x768@72 in a window, force full screen.  Otherwise, for now, remain windowed.  As we expand the interface to work better with more screen real estate, we may decide to go full screen on bigger screens to give a better experience.

phrack, 2016-09-15:
Based on this w3c data, it makes sense to set the minimum to 1280x800: http://www.w3schools.com/browsers/browsers_display.asp.

The actionable things here that get 3.9 out in a solid state and in the short term are:

- [ ] Maximize at 1280x800 or less, it's OK if it's not perfect looking at less but I'll make some effort to ensure it isn't complete garbage at 1024x768 for VMs if nothing else.
- [ ] The webcam feed tab container should get a slice of the window growth pie. Right now the training exercise area gets 100% of the height gain and the timer table 100% of the width gain. 25% in each dimension to the tab container seems like a reasonable starting point.

phrack, 2016-09-17:
Setting just the scaleX and scaleY on the anchorPane for the arena tab seems to do everything we need, which is nice and easy. Of course, it's JavaFX so fucky things happen to what you see in the tab when you do this. In particular, the canvas and such slides off to the side for seemingly no reason so you can't see all of it anymore.

commit 70ec10b8 (2016-09-17): Fit tabArenaPane to TabPane to minimize scrolling. Close #546
closed 2016-09-17 by phrack