phrack/ShootOFF #556: Oddities in camera registration

phrack, 2016-09-11:
Should IP camera registration use CameraFactory.registerCamera() since they are not discovered by webcam-capture? Shouldn't I be able to unregister a camera (this is certainly a thing for IP cameras)? Also, does the order the camera's are listed in really matter? It seems like the data structure would be a lot more efficient as a `HashSet`.

I am not entirely clear on how registered cameras are used so I didn't want ot mess with it myself.

cbdmaul, 2016-09-13:
The config class maintains a IP cam list because they have to be written out to the config, so they're registered/unregistered there.  My understanding is that they do show up in the Sarxos getWebcams list once they've been initialized? You would know better than I do.

In any case, the logic for IPcams could be mostly pulled out of configuration and our ipcamera class and put into CameraFactory.  Is it worth refactoring if it's rarely used and it'll probably break something?

phrack, 2016-09-14:
I was just confused because the method's comment isn't specific about what "discoverd by webcam-capture" means. IP cameras are not discovered, you tell webcam capture about them and then it repeats back to you in a list what you told it if you ask later. That made me wonder if I should have to register IP cameras too. 

I get now that this functionality is for cameras that have nothing to do with webcam-capture because webcam-capture doesn't and won't (for the foreseeable future) have any knowledge of their existence. That is, non-UVC cameras that also do not have a webcam-capture driver.

cbdmaul, 2016-09-14:
Yeah I could see how my comment was confusing.  I meant registered with webcam-capture.  It is worthy of refactoring but I don't think now is the right time.  If we break it we might not even notice.  When I started refactoring and tested ip cam, I already had to fix a bug to make them work again.

phrack, 2016-09-15:
I am going to change the comment and close this.

phrack, 2016-09-15:
This issue really should have been something about an AR-15. We'll have to remeber AK's for 762.

commit 187326ad (2016-09-15): Clarify CameraFactory to close #556
closed 2016-09-15 by phrack