phrack/ShootOFF #557: Concurrent Modification Exception

cbdmaul, 2016-09-13:
Exception in thread "oCam-5CRO-U 1" java.util.ConcurrentModificationException
	at java.util.HashMap$HashIterator.nextNode(HashMap.java:1429)
	at java.util.HashMap$KeyIterator.next(HashMap.java:1453)
	at com.shootoff.camera.shotdetection.JavaShotDetector.processFrame(JavaShotDetector.java:292)
	at com.shootoff.camera.CameraManager.processFrame(CameraManager.java:626)
	at com.shootoff.camera.CameraManager.handleFrame(CameraManager.java:515)
	at com.shootoff.camera.CameraManager.newFrame(CameraManager.java:498)
	at com.shootoff.camera.cameratypes.SarxosCaptureCamera.run(SarxosCaptureCamera.java:186)
	at java.lang.Thread.run(Thread.java:745)

phrack, 2016-09-16:
Why is this a `HashMap` error when 292 is reading a `Set` not a `Map`?

cbdmaul, 2016-09-16:
That's a very good question.  I thought that I might have had an indeterminate state, leading to a mistaken exception, but that line has been Set not Map for as long as I went back...

phrack, 2016-09-16:
Also,

1. It's coming from the oCam thread. Shouldn't JavaShotDetection be off if oCam is being used?
2. This is very battle hardened code, hasn't been changed in several versions and this never came up before.

I am inclined to close this and maybe open an issue about 1 if you think it's a bug.

cbdmaul, 2016-09-16:
The oCam is a regular camera, it uses JavaShotDetection

phrack, 2016-09-16:
In that case, we should give the thread a better name if we control it.

cbdmaul, 2016-09-16:
I changed the thread name to the camera name because if I had two cameras and one of the threads hung or crashed, I didn't know which camera it was.  We could add the detection method to the thread name as well.

phrack, 2016-09-16:
I'll do that in a second. I made a few other tweaks.

phrack, 2016-09-16:
PS. The problem is explained here: https://docs.oracle.com/javase/7/docs/api/java/util/Collections.html#synchronizedSet(java.util.Set). We just have to synchronize on the for loop. This should solve the problem given the way synchronizedSet wraps a normal set. I have a few other messaging tweaks, but nothing that's a big deal.

commit 5d78d2dd (2016-09-16): Clarify thread names and synchronized to iterate synchronizedSet to close #557
closed 2016-09-16 by phrack