phrack/ShootOFF #558: Shot To Click leaves visible mouse cursor

cbdmaul, 2016-09-13:
Ideally, Shot To Click either uses an invisible cursor or moves the cursor off screen after clicking

Don't know if this is only Windows behavior or if it's come up before.

phrack, 2016-09-14:
I don't think any window system gives you control of what the cursor looks like on someone elses window. We're lucky they even let us move it over a window that isn't ours and click.

If we filtered exercises by capabilities. Shot to click would firmly be on my IR + filter list.

cbdmaul, 2016-09-14:
Well that's why I mentioned just moving it out of the way after clicking.

phrack, 2016-09-15:
Seems like it'd be a simple update to just move the mouse back to where it was before automatically moving it. This, in theory, would cause the mouse to move around all over the place, but always go back to where the user left it when they started. I suggest trying that since you can actually test it. I honestly was never all that confident this exercise even worked decently since it cause so many motion issues.

closed 2016-09-16 by cbdmaul