phrack/ShootOFF #560: Move to installers rather than zips

cbdmaul, 2016-09-15:
Extracting the zip and then remembering where to go to run ShootOFF is a pain.  We should move to installers per OS.  Worth looking into a multiplatform installer toolkit.  I think this is worthy of inclusion with GUI improvements if it's not too difficult.  It is as much of a usability issue as the old GUI.

phrack, 2016-09-15:
This is what you're probably looking for: https://docs.oracle.com/javase/8/docs/technotes/guides/deploy/self-contained-packaging.html#A1324980. There was a lot of hype about this when JavaFX was new, but I haven't heard much about it since. Thus, I have no idea how well it works.

I suggest you play with it on Windows and see if it meets your expectation. We can upload multiple files per release, so uploading it along with the zip isn't a big deal. The complication release wise is that you can only generate releases for the OS you are doing the build on (i.e. don't expect to ever have an installer for anything but Windows even if you automate everything in the current gradle release process).

phrack, 2016-09-15:
And yes, I am booting a lot of this mundane crap to you because I've had my absolute fill of mundane crap for this release with the GUI. After #546 and #542 I want to do something fun.

cbdmaul, 2016-09-16:
I have dislocated limbs and had a more pleasant experience than working on this.  Will still try to tackle it soon.

phrack, 2016-09-17:
Yea, my natural instincts almost closed this issue before I even finished reading the title, knowing the tooling was likely long neglected and never fully developed. Probably why the hype train died because this would be soooo nice otherwise.

As another example, Scene Builder is no longer built by Oracle, but they do provide the source code with OpenJFX. This is legitametly the only source of new builds for Windows: http://gluonhq.com/labs/scene-builder/.

phrack, 2016-09-17:
Also, god help you if you end up actually trying to make this work in gradle. You'll start making arguments like, "Well, people who would get a hardware kit don't need an installer anyway" and you'll _know_ it's the most reasonable argument you've heard all day.

cbdmaul, 2016-09-17:
I followed this guide: http://code.makery.ch/library/javafx-8-tutorial/part7/

I actually have it really close.  Build is working, just need to get all of our resources inside it.

Not going to celebrate yet, and no Gradle integration, but I'd be satisfied.  Literally took me 10 minutes to get it this far.

cbdmaul, 2016-09-18:
Remaining issues:

1. There's no confirmation that the install finished successfully, and ShootOFF is not automatically launched.  This will probably a bitch to change, it would require generating our own WiX toolkit WSI file on the fly.

2. 64 bit only

3. Filename is missing "-final"

cbdmaul, 2016-09-19:
As far as I can tell, #3 is not fixable unless we manually rename the file with a gradle command.  Worth it or not?

phrack, 2016-09-19:
It's not worth it.

phrack, 2016-09-23:
Is this closeable and assignable to 3.9?

cbdmaul, 2016-09-24:
I think the solution to #2 is to build it with a 32 bit JDK.  So unless we have a dedicated build machine we're going to have to do that manually anyway.

#1 and #3 are a big enough pain to ignore for now.

closed 2016-09-24 by cbdmaul