phrack/ShootOFF #564: Checkboxes do not return after renaming a camera on preferences slide

phrack, 2016-09-17:


cbdmaul, 2016-09-20:
This only happens on Windows if you don't hit enter in the box fyi.

phrack, 2016-09-21:
I am getting this even if I hit enter on Linux. I think there is a bookkeeping issue with the camera names map.

phrack, 2016-09-23:
FUUUUUCK I HATE THIS GUI CODE

This code works as expected IF the camera is checked. The solution is to make it so that it's not editable unless  the checkbox is checked. The following code at the top of on mouse clicked handled for a list cell seems to do that well enough:

```java
				if (!fetchWebcamChecked(getText())) {
					this.setEditable(false);
					return;
				}
				
				if (event.getClickCount() > 1 && fetchWebcamChecked(getText())) {	
					this.setEditable(true);
					startEdit();
				}
```

The problem is that within the same session of ShootOFF the mapping between the renamed camera old camera name isn't maintained so it thinks the new name is a new camera when you re-open the preferences.

phrack, 2016-09-23:
This all so fucking obnoxiously overcomplicated because JavaFX gives you NO WAY to map a new item to a ListCell. You can't even say something simple like, "Hmm, I want the ListCell who's text is x" because obviously that'd be too convenient.

commit fc7fdef9 (2016-09-24): Don't allow camera renames for unconfigured cameras and fix camera renaming to close #564
closed 2016-09-24 by phrack