phrack/ShootOFF #566: Clay busters did not turn off when start calibration

phrack, 2016-09-18:
This is a general problem with exercise controls and calibrating.

cbdmaul, 2016-09-23:
Also, courses and other targets remain on the arena when clay busters is launched.

phrack, 2016-09-23:
That's expected. No exercise clears this for you and they don't even have the ability to do so. It's two button clicks to do this yourself, and if we add it to exercises it WILL get abused.

phrack, 2016-09-23:
What I am going to do here is if there is a set exercise when calibration starts, I am going to save which one is set, unset it, then reset it (meaning the exercise will start from scratch) when calibration stops. 

If you want to go farther in a separate issue, go head. I will still think it's unwise to let exercises clear courses, but I won't stop it.

commit 748a911e (2016-09-23): Save and disable projector exercise for calibration and restart it afterwards to close #566
closed 2016-09-23 by phrack