phrack/ShootOFF #571: JavaShotDetector OpenCV assertion error

phrack, 2016-09-18:
OpenCV Error: Assertion failed (0 <= _colRange.start && _colRange.start <= _colRange.end && _colRange.end <= m.cols) in Mat, file /mnt/hgfs/opencv/opencv/opencv-2.4.13/modules/core/src/matrix.cpp, line 292
Exception in thread "Shot detection Microsoft® LifeCam HD-3000 /dev/video0 com.shootoff.camera.shotdetection.JavaShotDetector" CvException [org.opencv.core.CvException: cv::Exception: /mnt/hgfs/opencv/opencv/opencv-2.4.13/modules/core/src/matrix.cpp:292: error: (-215) 0 <= _colRange.start && _colRange.start <= _colRange.end && _colRange.end <= m.cols in function Mat
]
	at org.opencv.core.Mat.n_submat_rr(Native Method)
	at org.opencv.core.Mat.submat(Mat.java:2271)
	at com.shootoff.camera.CameraManager.processFrame(CameraManager.java:603)
	at com.shootoff.camera.CameraManager.handleFrame(CameraManager.java:523)
	at com.shootoff.camera.CameraManager.newFrame(CameraManager.java:506)
	at com.shootoff.camera.cameratypes.SarxosCaptureCamera.run(SarxosCaptureCamera.java:182)
	at java.lang.Thread.run(Thread.java:745)

cbdmaul, 2016-09-19:
Think I'll need more debugging info to figure this out.  The projection bounds appear to be invalid.  Could just be failing to check for invalid coords somewhere.

cbdmaul, 2016-09-20:
Occurs to me that the received frame could be invalid / wrong size.  Added a check for that.

phrack, 2016-09-20:
I'll try it again later since I was reliably duplicating it. I was printing out the calibrated dimensions and they seemed perfectly reasonable.

phrack, 2016-09-21:
Closing this because it seems to be fixed.

closed 2016-09-21 by phrack