phrack/ShootOFF #574: camera window appears to be hidden by shot table when adding a new camera

ifly53e, 2016-09-20:
Not sure why, but there were times when I was testing with the new PS3Eye code and the camera window seemed like it was hidden behind the shot table.  I could hear the brightness and movement warnings but the camera pane was not there.  After I resized the window (sometimes vertically would do it, sometimes horizontally would do it) the camera pane would appear.

This was mostly when a new camera was added.  After the camera was saved to the preferences, this was not as reproducible.

cbdmaul, 2016-09-20:
I have seen the same issue.

phrack, 2016-09-21:
Please check to see if the fix for #565 fixed this. I fixed three different bugs that were causing #565 that could also have caused this.

phrack, 2016-09-22:
Anyone confirm? @ifly53e? @ beuller? @ beuller?

ifly53e, 2016-09-22:
Can't download the latest nightly for some reason...is it my work firewall?

phrack, 2016-09-22:
Might be, it's downloading at my house.

ifly53e, 2016-09-23:
After downloading the nightly and running for the first time with no external cameras and only the webcam on the laptap as the only cam, the camera window is still hidden until you collapse the main window fully horizontally...when you expand the window, the camera window is shown but not at full size.

phrack, 2016-09-23:
This bullshit only dupes on Windows and only if the screen resolution is >1280x800.

phrack, 2016-09-23:
This is fixed in my VM, thus I am going to make the wild and dangerous assumption that it is fixed everywhere.

closed 2016-09-23 by phrack