phrack/ShootOFF #576: ShootOFFController closes and reopens cameras unnecessarily in addConfiguredCameras

cbdmaul, 2016-09-20:


cbdmaul, 2016-09-20:
I would like to fix this but because there's no real link between the tabs and their cameramanager, it is a nightmare to identify what cameras have not changed status.

closed 2016-09-20 by cbdmaul

phrack, 2016-09-24:
You should try working with the listview if you think this is a nightmare. There is no link between a cell and the contents of the cell. You also cannot retrieve cells, yet cells are the only place to do things like change editability or detect when a change is occuring to its contents. This means you have to do all of your own bookkeeping in super hacky ways if you want to do things like rename an item in a cell or treat a cell differently if it's checked or not. On top of that, there are all kinds of wierd protocols, like you can't update the contents of a cell in a listener that tells you when the cell contents change. Don't like the change and want to make a small alteration? Too fucking bad. Add that all up and not only do you have to do the bookkeeping yourself, you have to do it yourself separately for essentially every little tiny feature you want to implement based on cell contents or states.

This of course results in very confusing code that ends up with strange bugs.

commit 19a69547 (2016-09-20): Fixes #576 by adding CameraManagerTabs in ShootOFFController.java

Fixes #572 by improving closure of Sarxos cameras

Possibly alleviates #571 by checking frame dimensions

Possibly alleviates #568 since ShootOFFController is smarter