phrack/ShootOFF #591: Using real world distances telltale chime causes thread not to close at shutdown

cbdmaul, 2016-09-24:
Seems like setting chime > telltale length has caused a rift in the space-time continuum

cbdmaul, 2016-09-24:
https://github.com/phrack/ShootOFF/blob/03470cea9f331c3b7b1d02b055c69ad1d178d69d/src/main/java/com/shootoff/gui/pane/ProjectorArenaPane.java#L598

https://github.com/phrack/ShootOFF/blob/537db60499c4eeb9826650cc011da7241a8763be/src/main/java/com/shootoff/gui/CanvasManager.java#L201

phrack, 2016-09-24:
This is indeed very weird. I also get it consistently only when the paper is detected to help debug it, this is what I do:

1. Open jvisual vm and double click the ShootOFF process.
2. Go to the Threads tab and sort by running. This shows you which threads are actually doing something.
3. Click Thread Dump so you can see the stack trace for each thread.

Doing this I found the watcher that looks for new plugins was still running, so I now correctly close that. However, that doesn't solve this problem. If you detect the paper it keeps running every time and the threads that are still going are Java threads, it's not even our code.

phrack, 2016-09-24:
Also weird. Calling shutdownNow should interrupt all of the waiting threads in a pool. The pool that plays the sound is indeed calling shutdownNow, but if you check if the thread was interrupted when the sound plays it says it was never interrupted.

phrack, 2016-09-24:
Figuring out the actual problem seemed hopeless. In this case the diagnostic message for finding the paper isn't supposed to play a chime anyway so I just ensured the future wouldn't even be created if the delay to play the chime is less than or equal to zero. This fixes the problem without figuring out what the heck is wrong with the pool.

commit 67e05ee6 (2016-09-24): Implement a proper way to not play a chime to close #591
closed 2016-09-24 by phrack