phrack/ShootOFF #599: logging calls are missing in the command line output on release builds

ifly53e, 2016-09-25:
Command line output says this...should it?

C:\Users\Tim\Desktop\nightly>java -jar ShootOFF.jar -d
SLF4J: A number (6) of logging calls during the initialization phase have been intercepted and are
SLF4J: now being replayed. These are subject to the filtering rules of the underlying logging system.
SLF4J: See also http://www.slf4j.org/codes.html#replay
2016-09-24 19:27:26 [JavaFX Application Thread] INFO  de.dfki.lt.tools.tokenizer.JTok - loading language resources for en from jtok/en
2016-09-24 19:27:27 [JavaFX Application Thread] INFO  de.dfki.lt.tools.tokenizer.JTok - loading language resources for en from jtok/en
2016-09-24 19:27:27 [JavaFX Application Thread] INFO  com.shootoff.Main - Error reporting has been disabled.
2016-09-24 19:27:28 [Plugin Watcher] DEBUG c.s.plugins.engine.PluginEngine - Starting to watch plugins directory
2016-09-24 19:27:28 [JavaFX Application Thread] WARN  com.github.sarxos.webcam.Webcam - No webcam has been detected!
2016-09-24 19:27:29 [JavaFX Application Thread] DEBUG com.shootoff.camera.CameraManager - starting camera thread PS3Eye

I think those six logging calls are missing in the output before the first INFO message

phrack, 2016-09-25:
Are those messages trace messages and you're in debug?

phrack, 2016-09-25:
PS. Here is an old example of setting just some of the autocalibration classes to TRACE while leaving everything else at its debug mode/non-debug mode defaults:

https://github.com/phrack/ShootOFF/blob/7683834c3d4279c3eb1de7b0ae1a59e1a1feced9/src/test/resources/logback-test.xml

You should do something similar to set the PS3eye code to TRACE. You will have to refresh the Eclipse project after doing this. DO NOT commit your logback changes.

I changed some of your log messages from debug to trace a few days go. My rule of thumb: if it's outputting the most critical state and results that sum of a large block of code or method call, go ahead and use DEBUG. If it's essentially just providing context about where in the code you are or very detailed state that is only useful when debugging very specific problems, use TRACE.

It's so easy to tune the log level for specific classes or packages in the logback.xml, as I show above, that we don't want to make the logs too noisy by putting a lot in debug.

closed 2016-09-25 by phrack

ifly53e, 2016-09-25:
Regular users running shootoff from the command line (not running eclipse) in debug and in regular mode are going to see those SLF4J messages.

phrack, 2016-09-25:
I automatically assume anyone running from the command line is more advanced and can deal with it or was asked to run from the command line and will be adding switches to add even more.

ifly53e, 2016-09-25:
I am with you...so lets assume you ask someone to run from the command line in debug mode.  You are not going to get the messages hidden by SLF4J because it is not replaying them.

phrack, 2016-09-25:
Yes, they will. The 6 replayed log messages are literally the 6 messages you showed above. I was explaining above that if you were expecting more messages, it's almost definitely because they are at TRACE.

ifly53e, 2016-09-25:
The first debug message in ps3eye in debug mode is not shown.

phrack, 2016-09-25:
Yes, because debug mode hasn't even been set yet. Debug mode isn't set until runShootOFF, which is called well after your initialization code.

ifly53e, 2016-09-25:
Well, from a layperson's perspective, if debug mode is set, debug messages should be presented, right?  I realize that noisy output can disrupt troubleshooting, but if the messages can show up when asked for, that would be good.  Can debug mode be set sooner in the start up process without breaking anything?

phrack, 2016-09-25:
There isn't a good solution here because your code is bootstrap code. Just like with an operating system, some services are not available yet if you are bootstrap code. The solution in this case if someone is having a problem is for them to edit logback.xml to manually configure the log level. There are a handful of problems that can happen with ShootOFF where this is the case.

ifly53e, 2016-09-25:
Ok...got it.  Thanks for the explaination.  Is logback.xml only available to users while editing the code in eclipse or can regular users edit logback.xml somewhere in the production code?

phrack, 2016-09-25:
I don't remember what the switch is off the top of my head, but there is a switch you use to tell logback to load its configuration from an external file. No regular user has run into a problem with bootstrap code so far, but if they did we would send them a log file, tell them where to put it, and give them a command line to copy and paste.

ifly53e, 2016-09-25:
ok...was just thinking through how to help troubleshoot the ps3eye if someone had a problem that the normal steps could not solve.  Thanks.

ifly53e, 2016-09-25:
"It can also be edited in the JAR file"
That's the ticket!  
In windows, I associated xml files to a word processor like notepad.
Then used WinRAR to open the shootoff.jar, double clicked on logback.xml, made the edits, hit save and winrar asked if I wanted to update the archive, said yes, and boom...edited the jar.
Thanks.