phrack/ShootOFF #602: Working with backgrounds

ifly53e, 2016-09-29:
There are times when only a certain portion of an image file is required for the entire background imageView.  Current methods to get the correct portion of an entire image involve the use of a photo editor and trial and error to zoom the image in or out and then crop the image.  Once the photo-shopped image has been found, it can be used for the ShootOFF background via the current interface.

To get around this issue, I used my "background" image as a target and brought it into the arena and stretched and zoomed the image to what I needed and then saved the whole thing as a course.  Then I load the course (which is just a zoomed in and stretched target) and bring in the rest of my exercise as normal which is just played over the course being used as the background.  

Using a course as a background is not working well with the new Arena tab.  When moving targets leave the arena via the top, bottom, or sides, the "Course/Background/Target", is moved/drifted as well.

I am trying to reverse engineer setDimensions in targetView and setCourse in ProjectorArena to make the background be positioned, scaled, and fitWidth/fitHeight correctly in CanvasManager's updateBackground.   I was trying to work backwards from the course's saved x,y,width,height of the target and translate it to the background imageView in updateBackground but I am lost in a quagmire of math inception and nothing I have tried has worked.  

So, I need some help and I am requesting a way to pinch and zoom an imported image graphically into ShootOFF to position a background as a background and not use my "course/background/target hack".  Or if the hack is the way to go, can the drifting target as a background be fixed so it remains in a constant position?

Thanks.

phrack, 2016-09-29:
The drifting background is because the canvas the targets and such are drawn on is a [group](https://docs.oracle.com/javase/8/javafx/api/javafx/scene/Group.html). Groups resize to contain all of their children and nothing else. What this means is that if a target moves off screen on the tab the group expands off screen and everything that isn't resizable shifts over to the new original to not be resized.

This isn't a problem on the arena window itself, at least not that I have seen.

Having identified that problem, I do not know the solution off the top of my head and find the properties involved confusing as well.

ifly53e, 2016-09-30:
Would it make sense to paint an underlying independent canvas with a target
from a course (or independant group) to be used as a background?

phrack, 2016-09-30:
It'd probably make things more confusing. I'd prefer to figure out how to keep the `Group` on the arena tab a consistent size and place regardless of where the targets it contains move.

ifly53e, 2016-10-01:
I am not sure if this makes sense, but is is possible to just draw the arena window twice, once for the projector, and once for the arena tab?  The only difference would be resolution and you seem to have the transformations down for translating everything from one size to another....just a thought (maybe a dumb one but oh well...)

phrack, 2016-10-01:
That was actually the original idea, and no it's not. JavaFX only allows a paintable component to have one parent, meaning it can only be drawn once.

phrack, 2016-10-01:
I played with this a bunch today and I am getting a better handle on it, but still not really there yet. What I see on the arena tab is generally reasonable, it's typically just showing me parts of the view that I can't see on the projector, which I think is fine. The big issue is that for some reason the background on the projector will get stretched when a target is moved off screen down or right. This should not be possible because the background on the projector is never supposed to be bigger than the window on the projector and that window is definitely not resizing when the stretching occurs.

If I can solve that stretching issue, I'll consider this closed. I am guessing, but too burned out to want to verify, that the `MirroredCanvasManager` mirrors `CanvasManager.setBackgroundFit` without respecting the invariant described above (that the background can't stretch beyond the size of the arena window).

commit 3c9aee25 (2016-10-02): Lock arena background fit to stage dimensions instead of pane dimensions to close #602
closed 2016-10-02 by phrack