phrack/ShootOFF #604: Oddities with perspective scaling

cbdmaul, 2016-10-01:
[image]

These two targets have identical parameters.  If I change height and change it back to 216, I get the size on the left.  If I change width and change it back to 279, I get the size on the right.

The one on the right (the bigger one) is the correctly sized one (Well, not perfect, but as correct as our math and measurement of the paper pattern is right now).

cbdmaul, 2016-10-01:
I would think this would be on the GUI side, as the perspective side is literally one function at that point and I don't see how you could possibly have undeterministic output from it.

cbdmaul, 2016-10-02:
This has to be in setDimension.  calculateObjectSize is always given the correct information and returns the correct value.

cbdmaul, 2016-10-02:
Yeah it has to do with what of the scaling functions in setDimension get called.  Changing height works as expected.  Changing width results in the too small target.  Any ideas?

phrack, 2016-10-02:
That's weird because the width calculation looks right but the height one is using `scaleX` where it's probably supposed to use `scaleY`: https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/TargetView.java#L246 . I'd expect height to be wrong and width to be right.

cbdmaul, 2016-10-02:
I probably confused the two.  if there's a bug that's probably the source.

commit da4cc5d3 (2016-10-02): Fix setDimensions height calculation to close #604
closed 2016-10-02 by phrack