phrack/ShootOFF #611: Camera thread not closing on exit

cbdmaul, 2016-10-02:
Don't know why I'm suddenly having this opencv issue but it happens every time with my c270 right now.  Camera works fine otherwise.

Thread [Shot detection Logitech HD Webcam C270 1 com.shootoff.camera.shotdetection.JavaShotDetector] (Suspended)	
	owns: SarxosCaptureCamera  (id=71)	
	VideoCapture.isOpened_0(long) line: not available [native method]	
	VideoCapture.isOpened() line: 268	
	SarxosCaptureCamera.isOpen() line: 131	
	SarxosCaptureCamera.run() line: 189	
	Thread.run() line: 745

cbdmaul, 2016-10-02:
Oddly, the arena window stays open (frozen)

phrack, 2016-10-02:
I am getting the same thing. ShootOFF doesn't close at all anymore, it just freezes on close.

phrack, 2016-10-02:
I synchronized a method that wasn't synchronized before because `closing` was guarded half the time but not in there: https://github.com/phrack/ShootOFF/commit/24241694ff2b990eaa65ca44c759c060fa269956. Might have broken something?

cbdmaul, 2016-10-02:
might be causing #609 also? never had that before either

phrack, 2016-10-02:
Yes, it's causing both. The right answer seems to be to make `closing` atomic instead.

commit 95c9cda5 (2016-10-02): Making camera closing flag atomic to close #611 and close #609 deadlocks
closed 2016-10-02 by phrack