phrack/ShootOFF #612: Plate not falling despite playing sound

cbdmaul, 2016-10-02:
[image]

See top left plate.  It made the hit sound but did not fall.  No errors.

Shooting it again made it fall.

phrack, 2016-10-02:
I can't make out the target. I faintly see an IPSC target, but I can't see what the steel one is.

cbdmaul, 2016-10-02:
[image]

Better? I tried to get an arena view of it but wasn't able to reproduce, despite reproducing it like 3 times before taking the first screenshot.

phrack, 2016-10-02:
I'll leave this open but I have no ideas what to do about it. I've read through the animation code dozens of times trying to track this down and never succeeded. There is one time I used to get it very regularly, but I could never trace it to anything in our code. I got a window system update shortly after and haven't seen it since.

My guess has always been it's a threading issue in the JavaFX window system. Whether we are triggering it or making it worse, I have no idea. This is right up there with canceling `ScheduledFutures` in things I've seen several times and have never been able to track down.