phrack/ShootOFF #619: Exception while playing clay busters for ~15 minutes

cbdmaul, 2016-10-08:
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException: -1
	at java.util.ArrayList.elementData(ArrayList.java:418)
	at java.util.ArrayList.get(ArrayList.java:431)
	at com.sun.javafx.collections.ObservableListWrapper.get(ObservableListWrapper.java:89)
	at com.sun.javafx.collections.VetoableListDecorator.get(VetoableListDecorator.java:306)
	at javafx.scene.Parent.updateCachedBounds(Parent.java:1591)
	at javafx.scene.Parent.recomputeBounds(Parent.java:1535)
	at javafx.scene.Parent.impl_computeGeomBounds(Parent.java:1388)
	at javafx.scene.layout.Region.impl_computeGeomBounds(Region.java:3078)
	at javafx.scene.Node.updateGeomBounds(Node.java:3577)
	at javafx.scene.Node.getGeomBounds(Node.java:3530)
	at javafx.scene.Node.getLocalBounds(Node.java:3478)
	at javafx.scene.Node.updateTxBounds(Node.java:3641)
	at javafx.scene.Node.getTransformedBounds(Node.java:3424)
	at javafx.scene.Node.updateBounds(Node.java:559)
	at javafx.scene.Parent.updateBounds(Parent.java:1719)
	at javafx.scene.Parent.updateBounds(Parent.java:1717)
	at javafx.scene.Parent.updateBounds(Parent.java:1717)
	at javafx.scene.Scene$ScenePulseListener.pulse(Scene.java:2404)
	at com.sun.javafx.tk.Toolkit.lambda$runPulse$30(Toolkit.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.Toolkit.runPulse(Toolkit.java:354)
	at com.sun.javafx.tk.Toolkit.firePulse(Toolkit.java:381)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:510)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:490)
	at com.sun.javafx.tk.quantum.QuantumToolkit.lambda$runToolkit$404(QuantumToolkit.java:319)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException: -1
	at java.util.ArrayList.elementData(ArrayList.java:418)
	at java.util.ArrayList.get(ArrayList.java:431)
	at com.sun.javafx.collections.ObservableListWrapper.get(ObservableListWrapper.java:89)
	at com.sun.javafx.collections.VetoableListDecorator.get(VetoableListDecorator.java:306)
	at javafx.scene.Parent.updateCachedBounds(Parent.java:1591)
	at javafx.scene.Parent.recomputeBounds(Parent.java:1535)
	at javafx.scene.Parent.impl_computeGeomBounds(Parent.java:1388)
	at javafx.scene.layout.Region.impl_computeGeomBounds(Region.java:3078)
	at javafx.scene.Node.updateGeomBounds(Node.java:3577)
	at javafx.scene.Node.getGeomBounds(Node.java:3530)
	at javafx.scene.Node.getLocalBounds(Node.java:3478)
	at javafx.scene.Node.updateTxBounds(Node.java:3641)
	at javafx.scene.Node.getTransformedBounds(Node.java:3424)
	at javafx.scene.Node.updateBounds(Node.java:559)
	at javafx.scene.Parent.updateBounds(Parent.java:1719)
	at javafx.scene.Parent.updateBounds(Parent.java:1717)
	at javafx.scene.Parent.updateBounds(Parent.java:1717)
	at javafx.scene.Scene$ScenePulseListener.pulse(Scene.java:2404)
	at com.sun.javafx.tk.Toolkit.lambda$runPulse$30(Toolkit.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.Toolkit.runPulse(Toolkit.java:354)
	at com.sun.javafx.tk.Toolkit.firePulse(Toolkit.java:381)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:510)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:490)
	at com.sun.javafx.tk.quantum.QuantumToolkit.lambda$runToolkit$404(QuantumToolkit.java:319)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException: -1
	at java.util.ArrayList.elementData(ArrayList.java:418)
	at java.util.ArrayList.get(ArrayList.java:431)
	at com.sun.javafx.collections.ObservableListWrapper.get(ObservableListWrapper.java:89)
	at com.sun.javafx.collections.VetoableListDecorator.get(VetoableListDecorator.java:306)
	at javafx.scene.Parent.updateCachedBounds(Parent.java:1591)
	at javafx.scene.Parent.recomputeBounds(Parent.java:1535)
	at javafx.scene.Parent.impl_computeGeomBounds(Parent.java:1388)
	at javafx.scene.layout.Region.impl_computeGeomBounds(Region.java:3078)
	at javafx.scene.Node.updateGeomBounds(Node.java:3577)
[... 48 lines cut ...]
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Thread.java:745)
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException
Exception in thread "JavaFX Application Thread" java.lang.ArrayIndexOutOfBoundsException

phrack, 2016-10-08:
We officially broke JavaFX.

phrack, 2017-02-18:
There are stack overflow answers now that suggests this is because we are not going something on the JavaFX thread that we should be. The fun part will be figuring out what:

http://stackoverflow.com/questions/38802707/random-arrayindexoutofboundsexception-in-javafx-application

phrack, 2017-02-18:
And this one that we may be setting a coordinate that makes JavaFX puke:

phrack, 2017-03-05:
This can be duped by changing the move time to every millisecond and the launch time to every half second. By doing this it took me 500 clays to dupe (the equivalent of running for a full hour with normal settings) but at this speed it doesn't take too long.

phrack, 2017-03-05:
This is a due to a JavaFX race condition that is easy to solve externally. The fix also makes the drawing a bit smoother because some of the artifacting we saw sometimes was the same race condition. I'll push out a fix shortly.

phrack, 2017-03-05:
Fixed here: https://github.com/phrack/ShootOFF-Clay-Buster/commit/b76ef8eeb2e37295b556f702a134e10e3583182a. I went ahead and smoothed out the arch a bit too so that they won't go to the side as much since this isn't realistic. I just released a new version of the exercise to the store.

closed 2017-03-05 by phrack