phrack/ShootOFF #621: Not Working with DroidCamX

ncagnoli, 2016-10-09:
Hi everyone.

My last setup is DroidCamX5 with Shootoff 2.7.x. But is not working with 3.9.

Can i help to fix it ?

Thanks for help!

phrack, 2016-10-09:
See: https://github.com/phrack/ShootOFF/wiki/FAQ#what-do-i-do-if-my-webcam-does-not-work

cbdmaul, 2016-10-09:
Have you tried adding it as an IP camera? https://stackoverflow.com/questions/36102437/streaming-droidcam-video-to-opencv-python-in-anyway-possible

ncagnoli, 2016-10-10:
Hi Friends! Yes i try this link phrack thanks.

I already have ShootOff working fine with a webcam and a projector. But working with the phone would be really cool!

My first attempt was in Windows 10. I will try in Archlinux and return news.

Thanks for help.

ncagnoli, 2016-10-20:
Yes yes!!

Worked perfectly witch Archlinux (OpenJDK JFX) with Galaxy Note 4 (Android 6.0.1) and DroidCamX 6.4.6.

In my noob opinion a great setup for dry training :)

I dont know why dont work in windows 10...but...is windows kkkkk

phrack, 2016-10-20:
OK, thanks for the update. Probably just a difference in drivers. I'll close this.

closed 2016-10-20 by phrack