phrack/ShootOFF #623: Section of screen not working on OSX

cbdmaul, 2016-10-12:
https://www.reddit.com/r/ShootOFF/comments/56tclr/shots_being_ignored_in_some_parts_of_arena/

Please attempt to repro on osx

phrack, 2016-10-15:
I am going to close this and attribute it to user error for now. I spent an hour on it and I couldn't find any issues to blame on anything but the MBP I used being slow as sin. I had to insert a few artificial delays into things like autocalibration to make everything work just because the laptop was so slow. Still, I actually got decent results considering FPS warnings. The whole screen worked for me.

cbdmaul, 2016-10-15:
did you try different feed resolutions? that was part of the thread on
Reddit.  I don't see how it could be user error...

phrack, 2016-10-15:
That could be part of the issue. Mac wants to see the camera to max resolution no matter what I do. If the calculations are done with the expected dimensions instead of the actual ones I suppose that could throw things off.

That'd be confusing to me though because it was putting the shot markers where the shots were after calibration.

phrack, 2016-10-15:
When the actual camera dimensions don't match what was requested, it actually just changes the feed dimensions to what the actual is:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/camera/CameraManager.java#L192

It seems like if anything the problem might be this todo:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/camera/CameraManager.java#L249

cbdmaul, 2016-10-15:
Don't think the to-do is the problem, since the resolution is set at
initialization in either case.  I don't have a great idea here, did you
check the sector disable code? maybe some of the check boxes are not being
set/unset correctly?

phrack, 2016-10-15:
Yea, all the sectors read on. Based on how the calibration pattern is drawn on Mac, I suppose it could be that their Macs are so fast they are calibration before the pattern is all the way up.

Since they are both bumbling around with the trace idea, I suppose I'll make a build that ignores shot sectors and just always detects everywhere and another with an artificial delay to guarantee to the pattern is fully up before trying to auto-calibrate. They can try both and report back.

cbdmaul, 2016-10-15:
I don't think a delay will change anything but you could make a build that
has trace set on auto calibration so we can get the calibration images.
there could just be a bug in my latest code there.  but his cropped image
looks correct

phrack, 2016-10-15:
I'll just try all three. I have as much faith the pattern drawing is the problem as the sectors (that is, no faith), but might as well definitively cross options off the list. The builds only take a few seconds anyway.

phrack, 2016-10-18:
Two good recent leads on this:

1. It happens even without the projector arena. 
2. It seems to only be searching the first 640x480 pixels of the frame even though Mac is forcing every camera to the max resolution: https://www.reddit.com/r/ShootOFF/comments/56tclr/shots_being_ignored_in_some_parts_of_arena/d8vss3b

I should be able to throw in some print statements on the very slow Mac I need to try to track this down a bit more since I don't actually need shot detection to work fast to poke at those facts.

phrack, 2016-10-18:
Fuck, that was hard. I got it at least for non-arena code. The problem is that the `PixelClusterManager` mantains its own feed width and height that never gets updated. It always things the resolution is 640x480 no matter what happens after it's initialized.

commit a55b59f0 (2016-10-18): Ensure pixel cluster manager feed dimensions get updated on feed size change to close #623
closed 2016-10-18 by phrack