phrack/ShootOFF #641: Shot times end up negative

phrack, 2016-10-23:
[image]

The first shot with a time of -0.01 is a real shot I made with a laser pointer. The shots with times that are large negative numbers are shot to click shots. I am assuming this is related to changes for the OptiTrack time stamps.

cbdmaul, 2016-10-24:
The problem is that an epoch time is maintained in CameraManager but the shot detectors return a full timestamp.  See cameraTimeToShotTime

I tried to make CameraManager handle that but I guess I failed.  The -0.01 makes sense (as a bug), if the system is slow, resetting the epoch will result in a time slightly slower than the initial shot.  The huge negative and positive ones, I don't know.

phrack, 2016-10-26:
I'd be OK closing this with just the -0.01 fixed since regular users will see that. Click to shot timestamps (the large negative ones) would be nice to fix eventually, but it doesn't hurt anything that I can think of.

cbdmaul, 2016-10-26:
In that case, all that needs to happen is make the reset take a timestamp parameter and set that as the epoch

phrack, 2016-10-26:
I am confused by the reset part because this happens even if there never was a reset. I consistently get it for the first shot after opening ShootOFF.

cbdmaul, 2016-10-26:
the epoch has to be set every time there is a first shot

commit 213523f4 (2016-10-26): Ensure frame timestamps start at 0 to close #641
commit 1f0b51a3 (2016-10-26): Merge pull request #644 from phrack/641fix

Fixes #641 and fixes #643.  Fixes click to shot
commit dc83f267 (2016-10-26): Fixes #641 and fixes #643.  Fixes click to shot
closed 2016-10-26 by phrack