phrack/ShootOFF #660: Ability to copy target images straight from internet/long range shooting

Andyg1984, 2017-01-09:
Is there the ability to copy in an image as a target straight from the internet instead of having to create a target? Also if usin targets sized for long range shooting i.e. 100 meters and further is there the ability to zoom in on the target after the shots have been taken to review shot placement.(I am using for rifle shooting practice rather than pistol).

phrack, 2017-01-11:
ShootOFF does not currently support adding a target straight from a picture on the Internet. This has never been considered because all of the targets we create as defaults require dimension changes, color changes, sound additions, etc. The vast majority of our users also don't create targets, thus we have never prioritized more advanced target creation features unless they directly benefit our own lives (i.e. this was the case with the feature to set the background in the target editor). Honestly, it takes less than a minute to make a target from a picture that already has appropriate colors and dimensions. I'd need to read more words to see what you're going for.

There is not a way to zoom in on a target to see precisely where the shot landed. Some people who make custom exercises for themselves have added this as a feature of their exercises, but this code has never been contributed to the canonical code repository (this one).

phrack, 2017-01-15:
Closing since no follow up in a week.

closed 2017-01-15 by phrack