phrack/ShootOFF #661: Laser POI configuration option for offset laser trainers

cbdmaul, 2017-01-11:
Add configuration options and an adjustment during the shot translation process that allows a user to specify an offset for shot markers based on their weapon's offset laser position.  The simplest option is to allow the user to specify this in pixels and rely on the user to make adjustments as needed.

More advanced suggestions are available in the reddit thread: https://www.reddit.com/r/ShootOFF/comments/5n9nr0/laser_poi_offset/

Suggestion by xanadu140

cbdmaul, 2017-01-15:
Sorry for the closures, I mistyped an issue number

phrack, 2017-03-01:
We got this again from someone else.

> Hi! I have different lasercartridges wich I use in Shootoff with different weapons. Pistol, carbine and some more. They all have the same problem, they are hard (impossible) to align when they are inserted in the chamber of the gun. When I use FATS and Lasershot there is a function wich help me to align in the software. I click on a button and a cross appears. I aim in the center of the cross and make a good shot. The software then remember my alignoffset and correct all my hits then when I´m shooting. Do you fantastic codehacker think this is possible to do in ShootOFF? Regards Fredrik

https://www.reddit.com/r/ShootOFF/comments/5wnmh2/aligning_the_laser_in_shootoff/

phrack, 2017-03-05:
https://www.reddit.com/r/ShootOFF/comments/5xmiua/possible_target_for_laser_aligning/

cbdmaul, 2017-03-05:
Any thoughts on implementation? It could be implemented as an exercise and we could provide a special hook to convey that information to the core.  Or we could use a autocalibration style mechanism, generalizing some of its hooks, to provide it in the core.

I'm leaning towards A and making it available in the exercise store.  Thoughts?

phrack, 2017-03-05:
It seems to me like the easiest thing that doesn't have bothersome implications is a special target tag. We could have a target meant specifically to do this with a new tag command to handle this. When the target gets shot the tag command is executed (https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/CanvasManager.java#L714) and the offset is saved in the config. Where the offset needs to be applied, if it exists in the config, apply it, otherwise move on.

We probably need to break that lambda out into its own class called `TargetCommands` or something. It doesn't really fit in the `CanvasManager` and it's really long for a lambda implementation.

cbdmaul, 2017-03-05:
Presumably the intent, like with the FATS setup, is that you shoot at all 5 targets.  Would that be a problem for that idea? Just average the shot location for each shot? Would need to know which one is the closest target, although I guess that information could be hard coded.

phrack, 2017-03-05:
I don't think it'd be a problem but you'd certainly need a single target with multiple regions (one per crosshair?) if you did that approach. I am not sure this is actually easier to do than an exercise, but it's less obnoxious than letting exercises change global configuration settings. I think they are of equal difficulty because know where a shot landed relative to something else has always been a pain with the JavaFX coordinate system.

I'll do that refactoring now so the command tag stuff with be in a file call `TargetCommands` in the same package as `TargetView`.

cbdmaul, 2017-03-05:
I see.  Regions make sense.  If we put the crosshair dead center in the region then it is easy to find the distance from the center of the region to the shot and use that for the alignment.

phrack, 2017-03-05:
Yea, that's exactly along the lines I was thinking in. I am just pushed the refactoring for target commands.

phrack, 2017-03-05:
PS. I think everything needed to implement this (including all required imports) lives here now:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/gui/targets/TargetCommands.java

In theory this is as easy as making a target and adding a POI tag to the cases.

cbdmaul, 2017-03-11:
Needs:

1. Testing
2. Should write to config? Commented out in Configuration.java right now
3. Better user interaction? (Show checkmarks or way to see offset while shooting poi target?)

cbdmaul, 2017-03-13:
Needs to account for arena<->feed resolution difference

cbdmaul, 2017-03-18:
Is there a way to make the target fill the arena? @phrack 

Also todo: add wiki page that explains usage

phrack, 2017-03-18:
Not without adding a new tag command whose implementation calls `target.setDimensions` with the results of `arenaPane.getArenaStageResolution()`.

cbdmaul, 2017-03-19:
I think #693 and some testing are all that are preventing this from being finished.

Careful attention to checking accuracy if target is resized (but with it filling the canvas, most people won't resize target anyway).

phrack, 2017-03-19:
Where is the wiki article that covers this? The wiki's home page hasn't been updated since Jan 11, I didn't see a new article in the right hand list (I could have missed it given the large number of articles though), and it wasn't added to any of the pages I guessed it might be under.

cbdmaul, 2017-03-19:
Didn't make it yet, was waiting for the fix on #693 before setting it up and taking some pictures.

cbdmaul, 2017-03-19:
https://github.com/phrack/ShootOFF/wiki/Point-of-Impact-Adjustment

phrack, 2017-03-19:
Seems to be close. Here is my problem: I aim for the center of the crosshair and my shots land very very close to each other. I do this until I get a different beep meaning POI has changed. Now my instinct is to want to test it so I keep the same point of aim and shoot again, which should put the new shot in the middle of the crosshair instead of being slightly off. I can't do this test though because it disables POI.

As a result I am having a bit of trouble deciding whether or not it's accurate. The code logic seems sound, I am just not sure the adjustment is actually reasonable and handling the scaling correctly.

phrack, 2017-03-19:
Also, assuming what I described is right, the wiki article should explicitly state to pick one of the crosshairs and take every shot aiming for its center.

cbdmaul, 2017-03-20:
I didn't specify about staying with a specific crosshair or moving because it doesn't affect the procedure, wasn't sure how to state that.

I have the same qualms about the procedure that you do but I am unsure what to do differently.

phrack, 2017-03-20:
Does it make sense to just drop to the middle crosshair if it's all the same? I've thought of a few reasons I might shoot at a crosshair that isn't the middle one but, those reasons seem fake and useless.

I selfishly haven't suggested the "ideal" approach because it won't work on Linux. Ideally we'd use `Target.addTargetChild`/`Target.removeTargetChild` to draw instructions on the target step-by-step through the process. Draw arrow pointing to the middle of a crosshair with big text saying, "Aim here and shoot." Perhaps with a counter for how many shots are left and finally ask for a test shot with a short description of what should happen.

This would be False Shot City on Linux and perhaps even Windows. A voice over wouldn't have this problem, but it presents a new challenge. Whatever is being said needs to be useful but also brief enough that I am not sitting there thinking, "Holy shit, shut up!" once I understand the procedure. In otherwords, I don't want it to be the shooting simulator version of the game NPC that won't shut up.

cbdmaul, 2017-03-20:
As long as the colors are dim we can probably get away with showing something to the user.

Ideally we would bring the same automatic exposure adjustment to linux....but...linux...

phrack, 2017-03-20:
OpenCV 3 might help with that problem assuming it doesn't require v4lcompat like OpenCV 2 does. I think a lot of camera oddities on Linux are due to the v4lcompat requirement. I may make this upgrade a centerpiece of 4.1 along with a few new features to the Android app.

cbdmaul, 2017-03-22:
We appear to still have a bug somewhere.  erpesjo reports having an offset of -43,-50 which should be near impossible (-50 is the maximum possible..to get -50 he should have to hit the edge pixel x 5)

phrack, 2017-03-22:
I duped it too while playing with it last night. I am not sure of the cause but it's obvious that about a quarter of the time the offset is way off.

cbdmaul, 2017-03-22:
if you can enable trace for TargetCommand it should show each offset being calculated, pre and post scaling.  The test doesn't cover that code so maybe a new test needs to be made to have better coverage.

cbdmaul, 2017-03-23:
So the problem is that we don't convert canvas to camera pixels.  Part of the problem is that TargetCommands "hit" x,y are translated to local coordinates in the target region, never know the proper coords in there.

So we need to know:
1. Where is the actual location of the crosshair in the feed?
2. Where is the actual location of the shot in the feed? (This is the same translation as the shot markers..)

The translation is available, the trick is getting the proper (useful) info into TargetCommands

Erpesjo, 2017-03-25:
A very simple idea is that we always put the cross in the center of the screen and then always move the offset to the center. I dont know if thats work in this type of programming, but I use that in other softwares. /Fredrik

cbdmaul, 2017-03-25:
Latest commit worked with click to shoot.  Needs real world testing.

Erpesjo, 2017-03-26:
[image]

Erpesjo, 2017-03-26:
Here you can see my hits. The hits in the center is from my offset. The i put a target in the arena and shoot at it. You can see my hits on the far right side.
shootoff.poiadjust.y=16.897737973867976
shootoff.poiadjust.x=323.5941006718273
[image]/Fredrik

phrack, 2017-03-26:
I too end up with a ginormous offset for 5 shots pretty close to the crosshair.

shootoff.poiadjust.y=42.488700174901226
shootoff.poiadjust.x=297.5954785111277

phrack, 2017-03-26:
I'll do this test with trace later. Trying to get out the door so I just wanted to make sure the result above duped at a minimum (it definitely did).

Erpesjo, 2017-03-26:
I dont know if you read my comment in the #reddit"
But my Reset button dont work. All shots remain at the screen. /Fredrik

phrack, 2017-03-26:
I duped the reset button bug. I wonder when the heck that broke. Trace log incoming as soon as I can get it to turn on trace...

phrack, 2017-03-26:
Shot 1:

2017-03-26 11:07:21 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (377.2941176470588, 291.5), cl RED fr 676
2017-03-26 11:07:21 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (632.9026036644165, 414.2857142857143), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-26 11:07:21 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI regcenterx 1201.311475409836 regcentery 502.7417027417027
2017-03-26 11:07:21 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI scalex 2.0 scaley 2.2222222222222223
2017-03-26 11:07:21 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI offsetx -284.2044358727098 offsety -39.80519480519478

Shot 2:

2017-03-26 11:08:22 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (385.55555555555554, 286.25925925925924), cl RED fr 2498
2017-03-26 11:08:22 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (654.5719489981784, 400.67340067340064), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-26 11:08:22 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI regcenterx 1201.311475409836 regcentery 502.7417027417027
2017-03-26 11:08:22 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI scalex 2.0 scaley 2.2222222222222223
2017-03-26 11:08:22 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI offsetx -273.3697632058288 offsety -45.93073593073592

Shot 3:

2017-03-26 11:08:53 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (373.92424242424244, 293.5984848484849), cl RED fr 3429
2017-03-26 11:08:53 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (624.0635866865375, 419.7363242817789), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-26 11:08:53 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI regcenterx 1201.311475409836 regcentery 502.7417027417027
2017-03-26 11:08:53 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI scalex 2.0 scaley 2.2222222222222223
2017-03-26 11:08:53 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI offsetx -288.62394436164925 offsety -37.35242030696571

Shot 4:

2017-03-26 11:09:20 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (371.625, 291.75), cl RED fr 4218
2017-03-26 11:09:20 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (618.0327868852459, 414.93506493506493), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-26 11:09:20 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI regcenterx 1201.311475409836 regcentery 502.7417027417027
2017-03-26 11:09:20 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI scalex 2.0 scaley 2.2222222222222223
2017-03-26 11:09:20 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI offsetx -291.6393442622951 offsety -39.51298701298699

Shot 5:

2017-03-26 11:09:50 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (374.94444444444446, 292.5925925925926), cl RED fr 5127
2017-03-26 11:09:50 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (626.7395264116575, 417.1236171236172), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-26 11:09:50 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI regcenterx 1201.311475409836 regcentery 502.7417027417027
2017-03-26 11:09:50 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI scalex 2.0 scaley 2.2222222222222223
2017-03-26 11:09:50 [Shot Notifier] TRACE c.s.gui.targets.TargetCommands - Adjusting POI offsetx -287.28597449908926 offsety -38.52813852813848

First shot after removing target:

2017-03-26 11:11:02 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (493.5246924403144, 253.42997694945743), cl RED fr 7279
2017-03-26 11:11:02 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Did Not Find Hit For Shot (937.7696850893492, 315.4025375310583)

phrack, 2017-03-26:
@Erpesjo I just pushed a fix for the reset but.

Erpesjo, 2017-03-27:
Nighly 03272017:
Reset button now works again. 
The offset is 100% right in side, but not in hight.
shootoff.poiadjust.y=-5.441929630794067
shootoff.poiadjust.x=2.8976253949654565
The hits in hight have not moved at all.
My laser hits is down to the left. And after the offset my hits on the screen in perfectly in side, but still low.
/Fredrik

cbdmaul, 2017-03-27:
It might just be #697 that is causing the placement to be wrong for you now.  Please test again after #697 is fixed.

Erpesjo, 2017-03-27:
I found another strange thing. If I choose to show shot markers on arena they appears in a very strange place. Not where i hit. They are shown far left and a little higher and more compact then they should.
/fredrik

cbdmaul, 2017-03-29:
#697 is fixed now.  I forgot to mention it in the commit, but I fixed a bug with loading the parameters from the config file as well.

Erpesjo, 2017-03-29:
Now the hits on the screen is placed where they should. (The are placed where the real laser hits with some side offset, but no hight offset)
I dont know if you have changed anything with the offset. But, the offset is still wrong in hights. In side it seems to be right. I will do some more testing next week. I wont have internet connection for a week I think. Have a nice weekend. /Fredrik

cbdmaul, 2017-03-29:
Can you enable arena markers in the preferences, take the 5 setup shots, and show us a screenshot or photo of the where your shots landed? And the output of what the offset was set to.

phrack, 2017-03-30:
I will plan to try it tonight now that I finally have an evening that is free.

Erpesjo, 2017-03-30:
Hi! I will after this weekend. I fly under the radar this weekend and have no internet until next week. Cheers Fredrik 

Njut av dagen, i morgon är det för sent.

phrack, 2017-03-30:
Seems to be right for me. I still get messed up by the test shot thing, but I am willing to let that go since this is a very niche feature anyway. 

Shot 1:

2017-03-30 17:28:40 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (367.5, 283.38709677419354), cl RED fr 334
2017-03-30 17:28:40 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (623.4262948207171, 405.7869012707722), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-30 17:28:40 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] DEBUG c.s.c.shotdetection.ShotDetector - Processing Shot: Shot Rejected By com.shootoff.camera.processors.DeduplicationProcessor

Shot 2:

2017-03-30 17:29:02 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (366.3271028037383, 289.7710280373832), cl RED fr 968
2017-03-30 17:29:02 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (620.4356406151096, 421.2630982724441), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-30 17:29:02 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] DEBUG c.s.c.shotdetection.ShotDetector - Processing Shot: Shot Rejected By com.shootoff.camera.processors.DeduplicationProcessor

Shot 3:

2017-03-30 17:29:19 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (364.2586206896552, 280.7844827586207), cl RED fr 1481
2017-03-30 17:29:19 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (615.1614232724276, 399.4775339602926), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-30 17:29:19 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] DEBUG c.s.c.shotdetection.ShotDetector - Processing Shot: Shot Rejected By com.shootoff.camera.processors.DeduplicationProcessor

Shot 4:

2017-03-30 17:29:39 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (367.96875, 284.09375), cl RED fr 2100
2017-03-30 17:29:39 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (624.6215139442231, 407.5), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-30 17:29:39 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] DEBUG c.s.c.shotdetection.ShotDetector - Processing Shot: Shot Rejected By com.shootoff.camera.processors.DeduplicationProcessor

Shot 5:

2017-03-30 17:29:57 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (364.140625, 299.453125), cl RED fr 2637
2017-03-30 17:29:57 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (614.8605577689243, 444.7348484848485), Type (RECTANGLE), Tags (name:center, visible:false, command:poi_adjust)
2017-03-30 17:29:57 [Shot Notifier] INFO  com.shootoff.config.Configuration - Setting POI Adjustment: x 4.37267765066066 y -4.037803431317778
2017-03-30 17:29:57 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] DEBUG c.s.c.shotdetection.ShotDetector - Processing Shot: Shot Rejected By com.shootoff.camera.processors.DeduplicationProcessor

First shot after end beep (at top left instead of middle):

2017-03-30 17:30:29 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] INFO  c.s.c.shotdetection.ShotDetector - Suspected shot accepted: Center (148.87267765066065, 149.96219656868223), cl RED fr 3578
2017-03-30 17:30:29 [Shot Notifier] DEBUG com.shootoff.gui.CanvasManager - Processing Shot: Found Hit Region For Shot (65.97017408933392, 82.33259774225995), Type (RECTANGLE), Tags (name:topleft, visible:false, command:poi_adjust)

Screenshot:

[image]

cbdmaul, 2017-03-30:
your results look reasonable to me, closing

closed 2017-03-30 by cbdmaul

Erpesjo, 2017-04-05:
Hi!
Just tested the new Nightly. The laser offset POI is still wrong i hight
and not perfekt in side.

I have tryed to change the POI manually in "shootoff.properties" but no
effekt in ShootOff software.
My POI-calibration shots:
[image: Infogad bild 1]
The result on a target:
[image: Infogad bild 2]
Still low hits to the left.
Regards
Fredrik

cbdmaul, 2017-04-06:
you did not appear to attach images correctly @erpesjo

commit ee384d4e (2017-01-15): Fixes #661, including test
commit c83fe0d1 (2017-01-15): Revert "Fixes #661, including test"

This reverts commit ee384d4e737e4d5dad56f05deee7dd7be484ed5e.