phrack/ShootOFF #662: OpenCV colrange exception on processFrame

phrack, 2017-01-14:
Exception in thread "Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector" CvException [org.opencv.core.CvException: cv::Exception: /mnt/hgfs/opencv/opencv/opencv-2.4.13/modules/core/src/matrix.cpp:292: error: (-215) 0 <= _colRange.start && _colRange.start <= _colRange.end && _colRange.end <= m.cols in function Mat
]
	at org.opencv.core.Mat.n_submat_rr(Native Method)
	at org.opencv.core.Mat.submat(Mat.java:2271)
	at com.shootoff.camera.CameraManager.processFrame(CameraManager.java:633)
	at com.shootoff.camera.CameraManager.handleFrame(CameraManager.java:551)
	at com.shootoff.camera.CameraManager.newFrame(CameraManager.java:512)
	at com.shootoff.camera.CameraManager.newFrame(CameraManager.java:507)
	at com.shootoff.camera.cameratypes.SarxosCaptureCamera.run(SarxosCaptureCamera.java:194)
	at java.lang.Thread.run(Thread.java:745)

phrack, 2017-01-14:
Here is the full log with trace for auto calibration:

2017-01-14 14:36:39 [JavaFX Application Thread] INFO  de.dfki.lt.tools.tokenizer.JTok - loading language resources for en from jtok/en
2017-01-14 14:36:39 [JavaFX Application Thread] INFO  de.dfki.lt.tools.tokenizer.JTok - loading language resources for en from jtok/en
2017-01-14 14:36:40 [JavaFX Application Thread] INFO  com.shootoff.Main - Error reporting has been disabled.
2017-01-14 14:36:41 [Plugin Watcher] DEBUG c.s.plugins.engine.PluginEngine - Starting to watch plugins directory
HIGHGUI ERROR: V4L/V4L2: VIDIOC_S_CROP
HIGHGUI ERROR: V4L: setting property #5 is not supported
2017-01-14 14:36:41 [JavaFX Application Thread] DEBUG com.shootoff.camera.CameraManager - starting camera thread Microsoft® LifeCam HD-3000 /dev/video0
libv4l2: warning v4l2 mmap buffers still mapped on close()
2017-01-14 14:36:50 [Plugin Watcher] DEBUG c.s.plugins.engine.PluginEngine - Stopped watching plugins directory
select timeout
2017-01-14 14:37:00 [Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector] WARN  com.shootoff.camera.CameraManager - Invalid frame yielded from Microsoft® LifeCam HD-3000 /dev/video0

@cbdmaul I'll email you the images.

phrack, 2017-01-14:
With the corner cut off trying to take video, I now get the exception earlier in the ACM:

OpenCV Error: Assertion failed (0 <= _colRange.start && _colRange.start <= _colRange.end && _colRange.end <= m.cols) in Mat, file /mnt/hgfs/opencv/opencv/opencv-2.4.13/modules/core/src/matrix.cpp, line 292
Exception in thread "Camera Microsoft® LifeCam HD-3000 /dev/video0 JavaShotDetector" CvException [org.opencv.core.CvException: cv::Exception: /mnt/hgfs/opencv/opencv/opencv-2.4.13/modules/core/src/matrix.cpp:292: error: (-215) 0 <= _colRange.start && _colRange.start <= _colRange.end && _colRange.end <= m.cols in function Mat
]
	at org.opencv.core.Mat.n_submat_rr(Native Method)
	at org.opencv.core.Mat.submat(Mat.java:2271)
	at com.shootoff.camera.autocalibration.AutoCalibrationManager.calibrateFrame(AutoCalibrationManager.java:566)
	at com.shootoff.camera.autocalibration.AutoCalibrationManager$StepFindBounds.process(AutoCalibrationManager.java:235)
	at com.shootoff.camera.autocalibration.AutoCalibrationManager.processFrame(AutoCalibrationManager.java:166)
	at com.shootoff.camera.CameraManager.processFrame(CameraManager.java:612)
	at com.shootoff.camera.CameraManager.handleFrame(CameraManager.java:552)
	at com.shootoff.camera.CameraManager.newFrame(CameraManager.java:513)
	at com.shootoff.camera.CameraManager.newFrame(CameraManager.java:508)
	at com.shootoff.camera.cameratypes.SarxosCaptureCamera.run(SarxosCaptureCamera.java:194)
	at java.lang.Thread.run(Thread.java:745)

cbdmaul, 2017-01-14:
Two issues

1. Bottom right corner is off the camera feed, but we don't catch it anywhere

2. When we warp perspective, we don't check to see if the perspective area is displayable.  We should instead see if we can make it fit in the screen area, if possible, and sacrifice anything else in the view to show the whole arena and calibrate successfully.  This might require enhancement instead of just a bug fix, so for now, just detect the condition.

commit e5f6202e (2017-01-15): Fixes #662
closed 2017-01-15 by cbdmaul