phrack/ShootOFF #668: Resize the shot marker based on real world distance data

cbdmaul, 2017-01-19:
Literally a shower thought.  The large shot marker has bugged me before.  Of course, it can't be too small (either you can't see it or the accuracy of the webcam isn't good to bother) but it would be nice if it wasn't huge on little targets.

phrack, 2017-01-20:
Could just never let it get smaller than can comfortably be seen on a high resolution display (i.e. have a minimum size). Could also make it standard-sized when clicked in the list?

This is NOT very common, but I've heard it a few times that people thought the shot marker size alters hit detection. We only use the center pixel for hit detection, but they thought if any part of a target was covered with the marker it would count as a hit. This is reasonable assuming there is a decent way to decide which target region was hit if the shot overlaps multiple regions.

cbdmaul, 2017-01-20:
I agree on both points, let the size fall inside a range that is visible at the resolution of the arena.  Make the shot bigger when it is selected.

For the hit area issue, we could approximate that by checking the four edges of the shot circle for target hits.  But without handling specific ammunition sizes I feel like any solution there is a hack more than a feature, in which case I'd rather put it off for a real solution.  That seems to be more of a ballistics / precision calculator feature set.