phrack/ShootOFF #680: glGetError 0x505 on Raspberry Pi

phrack, 2017-03-04:
glGetError 0x505
java.lang.NullPointerException
	at com.sun.prism.impl.BaseGraphics.drawTexture(BaseGraphics.java:411)
	at com.sun.prism.impl.ps.BaseShaderGraphics.drawTexture(BaseShaderGraphics.java:139)
	at com.sun.javafx.sg.prism.NGImageView.renderContent(NGImageView.java:123)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2056)
	at com.sun.javafx.sg.prism.NGImageView.doRender(NGImageView.java:103)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1948)
	at com.sun.javafx.sg.prism.NGGroup.renderContent(NGGroup.java:235)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2056)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1948)
	at com.sun.javafx.sg.prism.NGGroup.renderContent(NGGroup.java:235)
	at com.sun.javafx.sg.prism.NGRegion.renderContent(NGRegion.java:576)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2056)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1948)
	at com.sun.javafx.sg.prism.NGGroup.renderContent(NGGroup.java:235)
	at com.sun.javafx.sg.prism.NGRegion.renderContent(NGRegion.java:576)
	at com.sun.javafx.sg.prism.NGNode.renderForClip(NGNode.java:2297)
	at com.sun.javafx.sg.prism.NGNode.renderRectClip(NGNode.java:2191)
	at com.sun.javafx.sg.prism.NGNode.renderClip(NGNode.java:2217)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2050)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1948)
	at com.sun.javafx.sg.prism.NGGroup.renderContent(NGGroup.java:235)
	at com.sun.javafx.sg.prism.NGRegion.renderContent(NGRegion.java:576)
	at com.sun.javafx.sg.prism.NGNode.renderForClip(NGNode.java:2297)
	at com.sun.javafx.sg.prism.NGNode.renderRectClip(NGNode.java:2191)
	at com.sun.javafx.sg.prism.NGNode.renderClip(NGNode.java:2217)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2050)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1948)
	at com.sun.javafx.sg.prism.NGGroup.renderContent(NGGroup.java:235)
	at com.sun.javafx.sg.prism.NGRegion.renderContent(NGRegion.java:576)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2056)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1948)
	at com.sun.javafx.sg.prism.NGGroup.renderContent(NGGroup.java:235)
	at com.sun.javafx.sg.prism.NGRegion.renderContent(NGRegion.java:576)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2056)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1948)
	at com.sun.javafx.sg.prism.NGGroup.renderContent(NGGroup.java:235)
	at com.sun.javafx.sg.prism.NGRegion.renderContent(NGRegion.java:576)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2056)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1948)
	at com.sun.javafx.tk.quantum.ViewPainter.doPaint(ViewPainter.java:477)
	at com.sun.javafx.tk.quantum.ViewPainter.paintImpl(ViewPainter.java:323)
	at com.sun.javafx.tk.quantum.PresentingPainter.run(PresentingPainter.java:91)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at com.sun.javafx.tk.RenderJob.run(RenderJob.java:58)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:125)
	at java.lang.Thread.run(Thread.java:745)

phrack, 2017-03-25:
@cbdmaul I'll leave this here, but removed it from a milestone. This is a video memory issue we can maybe solve with OS configuration in Raspbian. If we can't find a sweet spot this may be the thing that moves us off the Pi.

cbdmaul, 2017-03-30:
Is anything else using video memory while ShootOFF is running? X?

phrack, 2017-03-30:
LXDE is too. I'll have to measure this. It's entirely possible JavaFX is just greedy with memory and will bloat into a small space quickly.