phrack/ShootOFF #684: ShootOFF-diagnostics.jar crashing...

zestysoft, 2017-03-06:
When opening ShootOFF-diagnostics,jar (v3.10-final), it starts fine, I can see my webcam and then after a few seconds I hear a sound and it disappears.

Here's the head of the log file:
`Localhost hostname lookup failed, keeping the value 'unavailable'
java.util.concurrent.TimeoutException: null
	at java.util.concurrent.FutureTask.get(Unknown Source)
	at com.getsentry.raven.event.EventBuilder$HostnameCache.updateCache(EventBuilder.java:620)`

Not sure what's broken -- my dns is working fine?  Is there an entry I can put in the hosts file to temporarily bypass this issue?

cbdmaul, 2017-03-06:
You just need to run the Shootoff-Diagnostics-Diagnostics.jar

My knowledge of this area is jokes only

phrack, 2017-03-06:
Looks like we need to update the Sentry libraries it is using. In any case, all it didn't do was send us shootoff-diagnostics.log, which can be done manually.

phrack, 2017-03-11:
I just put out a new version of the diagnostic tool with the newest Sentry libraries:

https://github.com/phrack/ShootOFF-diagnostics/releases/tag/v1.3-FINAL

I also checked that instance of Sentry is running fine. That seems to be everything checkable on our end. 

If you continue to have problems, it is using HTTP to communicate with dev.shootoffapp.com:9000.

closed 2017-03-11 by phrack